prusa3d/PrusaSlicer #549: Sequentiel preview is wrong

BeNeDeLuX, 2017-10-21:
### Version
Slic3r 1.37.1-prusa-3d-win64

### Operating system type + version
Windows 10 x64 / 1703

### Behavior
The preview for a sequentiel print is wrong. The models on the bed are "all at once" in the preview instead one after one.

Here is the generated gcode (if needed):
[xyzCalibration_cube_seq.zip](https://github.com/prusa3d/Slic3r/files/1404421/xyzCalibration_cube_seq.zip)

Please check the Prusa Forum for more infos and screenshots --> https://shop.prusa3d.com/forum/print-tips-slic3r-settings-kisslicer-model-repair--f12/slic3r-print-sequentiel-preview-is-wrong-t6240.html

bubnikv, 2017-11-30:
I see now. There is a misunderstanding. The print path preview in Slic3r does not show the G-code, but some intermediate data, where the path ordering, tool ordering for multi material prints, speeds and accelerations are not known yet. Your issue is similar to:

https://github.com/prusa3d/Slic3r/issues/23
https://github.com/prusa3d/Slic3r/issues/112

We plan to do a proper print path preview.

bubnikv, 2018-02-12:
Closing this issue. By the way, we have a G-code preview in the making.

closed 2018-02-12 by bubnikv