prusa3d/PrusaSlicer #550: Please keep SLIC3R_VERSION up to date when making new releases

hyperair, 2017-10-22:
### Version

1.33.8-devel - 1.37.1

### Operating system type + version

Ubuntu 17.04

### Behavior

* Slic3r Prusa thinks that it's at 1.33.8-devel and shows that value all over the place (about dialog, gcode exports, etc.)

See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=878014 for more information.

bubnikv, 2017-11-28:
Normally the build number is being generated by our internal build system by running "git describe". I would certainly forgot to change the particular header file manually at each release.

Would it solve your issue, if I just replace the version number in the particular header file by the cmake build script calling "git describe" to deduct the version?

bubnikv, 2017-12-05:
I will try to remember to bump up the version number at each release manually. Thanks for your reminder. Closing.

closed 2017-12-05 by bubnikv

hyperair, 2017-12-14:
Sorry, must have missed your initial reply. It would help if you do that `git describe` step as part of the tarball release process, and encoded its value into a file that's then included during the build step. The .git directory isn't present during a build from the source tarball, so `git describe` won't work for building Debian packages.

bubnikv, 2017-12-14:
Thanks for the response. The tarballs are generated by github automatically, I don't have any control over it. I will simply try to remember and bump up the version manually. That's not the best solution, but not such a big issue for me anyway.