prusa3d/PrusaSlicer #551: Many objects added, One not Manifold, Warning is displayed on all

Panayiotis-git, 2017-10-24:
### Version
1.37.1-prusa3d-win64

### Operating system type + version
Windows 7 64-bit

### Behavior
Add more than one or two objects. One of them must have some Manifold errors. If we select this one, then at the right-bottom corner, at the info group, is shown the warning sign that popups the errors detected.
If then we click on any of the others, even if the Manifold is yes, the warning icon is still present. It shouldn't.
Only if we deselect all objects the warning will be hidden. It should be hidden if the selected object is Manifold.

The uploaded stl file is not Manifold.
[HotEndCover_v0.4.zip](https://github.com/prusa3d/Slic3r/files/1412605/HotEndCover_v0.4.zip)

This image is correct behaviour:
[image]

This image is for a manifold object and the sign should be hidden:
[image]

bubnikv, 2018-03-15:
confirmed by @FidelCapo on 1.37.2 and our latest integration build 1.40.0-alpha.251.

bubnikv, 2018-03-15:
@enricoturri1966 Would you please look into this issue? Thanks.

FidelCapo, 2018-08-03:
This bug was fix in version 1.40.0. Closing

closed 2018-08-03 by FidelCapo