prusa3d/PrusaSlicer #552: Reload from Disk issue

webhive, 2017-10-25:
### Version
1.37.2

### Operating system type + version
Linux 4.13.7-1-ARCH #1 SMP PREEMPT Sat Oct 14 20:13:26 CEST 2017 x86_64 GNU/Linux

### Behavior
* 'Reload from Disk' action break existing model rotation
* _Steps needed to reproduce the problem_
  * Add model to plate
  * Rotate it anyway
  * Right lick on model
  * Press 'Reload from Disk' menu item 
* _Expected Results_
  * Updated model, but preserved rotation
* _Actual Results_
  * Model updated but rotation reverted to original

_Is this a new feature request?_
No

Ziethriel, 2017-10-25:
I have a similar issue with "Reload from Disk" on windows, probably related to the same issue. When reloading a scaled up mesh from disk it looks correct in 3d view, but will only slice up to the height of the unscaled object.

Example:
1) Scale a 1" cube to 1000% to get 10" cube.
2) Reload from disk
3) Slice and preview
4) Layers are correctly sliced to scale, but only to the height of 1"

bubnikv, 2017-11-30:
I just merged this feature from the original slic3r with the highest hopes. It seems though that this feature is half baked.

If you reload an object, you will lose the modifier meshes as well.

bubnikv, 2017-11-30:
Also please note that only rotation around the Z axis and scaling is maintained when the STL is reloaded. As of now, the rotation around any other axis, or non-uniform scaling is applied to the mesh directly, so this kind of manipulation is being lost on reload.

bubnikv, 2019-12-18:
Reload from disk has been reworked for the upcoming PrusaSlicer 2.2. Closing.

closed 2019-12-18 by bubnikv