prusa3d/PrusaSlicer #553: Support Generates Variable Layer Height - No Way to Disable

Pondering133, 2017-10-25:
### Version
1.37.2-prusa3d-win64

### Operating system type + version
Windows 10 Pro 64-bit, version 1703, build 15063.674

### Behavior
* On a single extruder machine printing at 0.2mm layer height and 0.4mm nozzle, supports are being generated at variable layer heights upwards of 2.00mm thick (My extruder cannot keep up with that volume flow). There does not seem to be an option to disable variable, or at least limit the maximum layer height of the variable supports. Option to "Synchronize with object layers" is greyed out and cannot be selected. Disabling "Enable Variable Layer Height Feature" on the Machine settings tab has no effect on the generated supports.

* _Steps needed to reproduce the problem_
  * Load model
  * Set to 0.2mm layer height
  * Enable support generation
* _Expected Results_
  * Supports generated at the model layer height of 0.2mm.
* _Actual Results_
  * Supports are generated with variable layer heights from 0,2mm-2.00mm.

#### STL/Config (.ZIP) where problem occurs
* Every file I've tried, small example attached: [GUST_shroud.zip](https://github.com/prusa3d/Slic3r/files/1415087/GUST_shroud.zip)
* [image]
* [image]
* [image]

bubnikv, 2017-10-25:
Slic3r does what you asked it for:

[image]

Pondering133, 2017-10-25:
Admittedly I missed that Max setting. However, if I set it to 0.3mm Machine Max and set Print Layer Height to 0.15mm, the supports are still being generated at variable heights 0.15-0.3mm. I cannot disable variable height supports.

bubnikv, 2017-10-25:
>  I cannot disable variable height supports.

Is it a problem?

Pondering133, 2017-10-25:
With Machine Max set to 0.2mm, Machine Min set to 0.2mm, printing at 0.2mm, there are now 0.1mm support layers being generated. The Machine's variable height is disabled. Sync Layers is greyed out.

bubnikv, 2017-10-25:
> With Machine Max set to 0.2mm, Machine Min set to 0.2mm, printing at 0.2mm, there are now 0.1mm support layers being generated. The Machine's variable height is disabled. Sync Layers is greyed out.

What is your "Contact Z distance"? Is it possible, that Slic3r fulfills all the constraints at once? Is it possible to create your configured Contact Z distance with a min/max layer height set to 0.2?

Is it a problem anyway?

Pondering133, 2017-10-25:
Contact Z is still default "0.2mm (detachable)". Changing it manually to just "0.2" still recreates the 0.1mm supports.

Well this is to report bugs/issues correct? This causes extra print time on supports, beyond the minimum height limit set by the machine settings, and against the setting that allows variable layers.

Itox001, 2017-11-23:
I think that it actually is a problem for some machines, since some Z setups are not as robust as the original Prusa with the integrated leadscrews on the motors, and repeatability might not be as good. This results in uneven layers when it has to stop in between the object's layers to extrude the support's layers.

DocYester, 2020-05-10:
Is there a chance that the option "Synchronize with object layers" will get usable at some time ?

tonyshackelton, 2022-07-22:
Still very broken, superslicer is too. Didn't really effect my prints until I switched nozzle's to a larger size, now I can't print anything with supports... the Z going up and down during a print just makes a huge mess. (yes I've fiddled with min/max, min is also ignored as shown by others above)

PS - If it's not a intended to be a global var, move it out of global part of the GUI ;) Printer settings should effect the.. Umm printer, not just part of a print.

tonyshackelton, 2022-07-22:
> What is your "Contact Z distance"? Is it possible, that Slic3r fulfills all the constraints at once? Is it possible to create your configured Contact Z distance with a min/max layer height set to [...]

Yes it's a problem.... A global variable that isn't global, is a bug (or move it to the appropriate place in the GUI). Min and Max are set the same, and supports just do whatever anyways, is a bug. (set everything for .2, get .125 supports) It wasn't a problem for me at all until I moved from a .4 to .5 nozzle. I haven't had a single good print with supports in PS/SS since. I'm glad it not effecting you....but it is certainly a problem for many, many users. Try a funny sized nozzle, the undisableable support layer height logic goes bonkers and just does whateva. It's a good time with a glass of wine, when you're here, you are family.....

closed 2026-09-29 by prusaslicer-bot[bot]