prusa3d/PrusaSlicer #554: CCaught C++ exception of type or derived from 'std::exception': vector::_M_range_check: __n

quirxi, 2017-10-26:
### Version
Version 1.33.8.devel
$ git describe --tag
version_1.38.0-1-g7308017e

### Operating system type + version
$ uname -a
Linux debian 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u5 (2017-09-19) x86_64 GNU/Linux
$ cat /etc/debian_version 
9.2

### Behavior
* Whenever I try to add support to a certain model and fiddle around with the various support settings in the "Print Settings"->"Support material" menu after a certain time slic3r crashes with a segmentation fault, or it throws an exception:
```
# ./slic3r
12:02:02: Debug: Unrecognized accel key ' del', accel string ignored.
12:02:02: Debug: Unknown accel modifier: ' '
12:02:02: Debug: No accel key found, accel string ignored.
12:02:02: Debug: Unknown accel modifier: ' '
12:02:02: Debug: No accel key found, accel string ignored.
12:02:02: Debug: Unrecognized accel key ' l', accel string ignored.
12:02:02: Debug: Unrecognized accel key ' r', accel string ignored.
12:02:02: Debug: Unrecognized accel key ' s', accel string ignored.
12:02:02: Debug: Unrecognized accel key ' 0', accel string ignored.
12:02:02: Debug: Unrecognized accel key ' 1', accel string ignored.
12:02:02: Debug: Unrecognized accel key ' 2', accel string ignored.
12:02:02: Debug: Unrecognized accel key ' 3', accel string ignored.
12:02:02: Debug: Unrecognized accel key ' 4', accel string ignored.
12:02:02: Debug: Unrecognized accel key ' 5', accel string ignored.
12:02:02: Debug: Unrecognized accel key ' 6', accel string ignored.
Error creating proxy: Verbindung ist geschlossen (g-io-error-quark, 18)
Error creating proxy: Verbindung ist geschlossen (g-io-error-quark, 18)
Error creating proxy: Verbindung ist geschlossen (g-io-error-quark, 18)
Error creating proxy: Verbindung ist geschlossen (g-io-error-quark, 18)
Error creating proxy: Verbindung ist geschlossen (g-io-error-quark, 18)
Caught C++ exception of type or derived from 'std::exception': vector::_M_range_check: __n (which is 4) >= this->size() (which is 0) at /media/arno/funtoo/home/arno/PROJECTS/3D/software/Slic3r/Slic3r_Prusa_Git/Slic3r/lib/Slic3r/Print/Object.pm line 24.
Perl exited with active threads:
	1 running and unjoined
	0 finished and unjoined
	0 running and detached
$
```

* _Steps needed to reproduce the problem:
1. Open slic3r
2.  Load the attached configuration file
3. Open the attached .stl
4. In "Print Settings"->"Support material" try to get the support right. After some time it crashes. But as far as I know not always at the same point.

* Expected Results: Support for model should be generated.
* Actual Results: Slic3r throws an exception and crashes.

[slic3r_error.zip](https://github.com/prusa3d/Slic3r/files/1417946/slic3r_error.zip)

bubnikv, 2017-10-26:
Thanks for the report.

Please note that the github master is generally unstable. For production, you should use the latest stable release, which is 

https://github.com/prusa3d/Slic3r/releases

quirxi, 2017-10-26:
In fact, the problem showed up on the stable release i downloaded from here. Just to make sure that the bug was not already fixed I compiled the newest version from git before i reported it. 
Btw.: Thanks for the cool program - I like it very much :)

bubnikv, 2017-11-06:
Thanks for the report.
I just wonder, how did you get this kind of exception, as the vector range checking is supposed to be disabled in a release build. So technically there is a bug, only it should not manifest this way.

Do you get this exception from a stable build? Where did you get the binaries from? Or did you compile the build yourself?

quirxi, 2017-11-06:
I actually got this kind of exception in two different versions:

1. Version 1.33.8.devel ($ git describe --tag: version_1.38.0-1-g7308017e)
2. Version: Slic3r-1.37.1-prusa3d-linux64-full-201709141215

The first time i got it it was in Slic3r-1.37.1-prusa3d-linux64-full-201709141215 I downloaded from __https://github.com/prusa3d/Slic3r/releases__. 
Just to make sure that I would not report a bug that already had been fixed I cloned the github repository (https://github.com/prusa3d/Slic3r.git) and installed it from sources and the same bug showed again. It only happend when i tried to add support to the inside of a kind of pot with holes in the walls.

bubnikv, 2018-08-02:
This bug is stalled, and we were not able to reproduce it. Let's see whether it will show up in the current Slic3r releases. Closing.

closed 2018-08-02 by bubnikv