prusa3d/PrusaSlicer #555: Support problems with models with hole in the walls.

quirxi, 2017-10-27:
### Version
Slic3r-1.37.1-prusa3d-linux64-full-201709141215

### Operating system type + version
$ uname -a
Linux debian 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u5 (2017-09-19) x86_64 GNU/Linux
$ cat /etc/debian_version
9.2

### Behavior
* I load a model of a "net pot" that is basically a pot with holes in its wall. I rotate the model so that the bottom of the pot is in the air because so i need less support and thus printing time. When I try to add the support ("Print Settings"->"Support material") with the option "Don't support bridges:" checked it only tries to support the bridges in the wall, but not the top of the model. When option "Don't support bridges:" is uncheked it supports the bridges in the wall and the top of the model it is supposed to support. It looks like it could not decide what is a bridge and what is the wall of an model its supposed to support. I tried also Cura and it had no problems with that model. Please have a look at the attached pictures where probably can see better what I try to describe here. Slic3r also often crashes when I try out the support settings as already reported in that issue: https://github.com/prusa3d/Slic3r/issues/554

1. Steps needed to reproduce the problem
2. Open provided .stl file.
3. Load provided config.ini file.
4. Check and uncheck the option "Don't support bridges:" in the menu "Print Settings"->"Support material".
5. See the result in the "Plater"->"Preview" window.

* _Expected Results_
Support should be generated properly.

* _Actual Results_
Support is not generated properly.

#### STL/Config (.ZIP) where problem occurs
[Slic3rSupportError.zip](https://github.com/prusa3d/Slic3r/files/1423355/Slic3rSupportError.zip)

Hugoz12, 2017-10-28:
Try increasing the 'XY separation'. In Cura you have set it to 2mm. When I try setting it to 2mm it works for me with 'Don't support bridges' unchecked.

quirxi, 2017-10-28:
Thank you for the tip ! :) 
It seems to work partly. 
When I use the honeycomb pattern with xy-seperation of 2 mm, pattern spacing of 4 mm and unchecked 'Don't support bridges' I still get strange artifacts outside of the object. And in this setting the 'Pillar' and the 'Honeycomb' support pattern are undistinguishable!?
With a xy-seperation of 3 mm the artifacts are gone but still no difference between 'Pillar' and the 'Honeycomb' support pattern!? Also with an pattern spacing of 4mm and the 'Rectilinear Grid' pattern I get these strange artifacts outside the perimeters walls.

[splic3r_support_pattern.zip](https://github.com/prusa3d/Slic3r/files/1424574/splic3r_support_pattern.zip)

bubnikv, 2021-03-22:
Testing snug supports

[image]

[image]

[image]

bubnikv, 2021-03-23:
Fixed with af9c7c9

    Implementing a new switch for the shape of support towers:
    expanded to a grid (the old way) vs.
    snug (like the upstream Slic3r, Cura or Ideamaker).
    
    Snug supports suffered from the degeneracies when merging overhang islands
    over a large number of layers when projecting the support towers down.
    We borrowed the idea & a bit of code from Cura by simplifying the support
    polygons by closing the concave cracks, see the smooth_outward() function
    and the MutablePolygon class.

closed 2021-03-23 by bubnikv