prusa3d/PrusaSlicer #556: Request - Wipe Tower per-filament settings

gregspruce, 2017-10-28:
Per-filament settings in the wipe tower could help dramatically reduce waste and reduce print time.

For example, Fillamentum Nobble Blue needs significantly more purging than Rapunzel Silver to prevent bleeding into the next color in the model. If we could specify that Extruder 4 needs a depth of 20, while the rest need a depth of 10, that would reduce time and filament spent in the tower portion.

bubnikv, 2017-11-06:
that is another one on my list

I know Peter wrote his own post processor, providing richer configuration options.
http://www.kisslicertalk.com/viewtopic.php?f=27&t=1988

cosmith71, 2017-11-12:
This is #1 on my wish list.  There are some filaments I won't use simply because of the wipe required.  Being able to adjust them individually would be great!

gregspruce, 2017-11-22:
> that is another one on my list

I know Peter wrote his own post processor, providing richer configuration options.
http://www.kisslicertalk.com/viewtopic.php?f=27&t=1988

I recently bought KISSlicer Pro, in large part for Peter's post processor and also for their automatic variable layers. Slic3r would benefit greatly from both of those features IMO.

bubnikv, 2017-11-23:
We have a new hire, Lukas Matena. He is now getting up to speed with the wipe tower code, he will try to integrate Peter's work into Slic3r.

cosmith71, 2017-11-23:
That's good news!  Welcome, Lukas.

srepmub, 2017-11-29:
Yes, please! I love the MMU so far, but printing with common black is really painful..

Blizzke, 2018-02-11:
Yes please! With white in it I have to use at least 25mm to avoid color bleeding and that makes the print time a whole lot longer!
[image]

lukasmatena, 2018-05-10:
This has been implemented as part of the changes done on the wipe tower code recently. The amount of wipe can be adjusted for each filament pair separately, the ramming can be customized, wipe tower rotated, etc. Also, variable layer height is now enabled for MM prints.

The changes are already merged in the current master, public release 1.40.1-alpha is expected in several days. Please, give us feedback after you have the chance to test it.

I'm closing the issue now.

closed 2018-05-10 by lukasmatena

ichbinscheisse, 2018-07-04:
Where do I find per color change depth in 1.40.1?

bubnikv, 2018-07-04:
[image]

ichbinscheisse, 2018-07-05:
Is there a reason why these settings can't be found in Multiple Extruders->Wipe Tower settings?
I can't access 'purging volumes.." because of using alot of virtual extruders and slic3r won't let you scroll down.
[image]

ichbinscheisse, 2018-07-05:
@Blizzke 
Diamond Hotend with Repetier firmware.
Repetier only allows for 16 virtual extruders, so you to overwrite mixing ratios of existing extruders via postprocessing to get more mixing ratios.
Slic3r and Slic3rPE let you define as many tools as you want fortunately...otherwise you would have a hard time replacing those tool change commands with the correct mixing ratio.