prusa3d/PrusaSlicer #557: No way to disable "prime multiextruder at edge of print surface" - causes problems for non-Prusa printers

ex-nerd, 2017-10-29:
### Version
1.37.2

### Operating system type + version
MacOS 10.13

### Behavior

First and most importantly, there does not seem to be a way to disable the multi-extruder "prime at edge of print bed" functionality added in 1.37.1.  There are multiple issues because of this.  For my Duet-controlled delta printer with a custom priming setup, this means:

* The priming tower begins to print without (re)setting a Z height, so if there is any custom Start G-code (e.g. a custom priming squirt off the side of the bed that wipes the nozzle on the bed that leaves the nozzle 5mm above the print surface), the prime tower begins printing in mid-air
* In the case of my printer, there are bed retention clips in the way of the nozzle at the location where the prime tower is supposed to print, so even if the Z height issue is addressed, I would get a head crash while trying to print the prime tower).  In other words, please make this configurable.
* The `M1 Remove priming towers and click button.` message is not interpreted the same on all printers.  In the case of the Duet (https://duet3d.com/wiki/G-code#M1:_Sleep_or_Conditional_stop), it basically does nothing because the gcode generated by slic3r has the `M1` immediately followed by a `G1` that wakes it back up again.
  * This also means that I'm left with a priming tower (assuming the problems mentioned in the previous bullet point is fixed) potentially in the middle of where I want to print because there wasn't time for me to get in and clean it off.
* If using a wipe tower (not to be confused with the "prime tower" mentioned above"), there appears to be a visual glitch in Preview for the first layer, showing an odd structure going off along the X axis: https://www.dropbox.com/s/4v90nikoj6hzzkk/prusa%20slic3r%20visual%20glitch.png?dl=0
  * Because of the issues mentioned earlier, I can't confirm if this is related or separate.  It's not present in 1.36.2, so I include it here for completeness.

#### STL/Config (.ZIP) where problem occurs
Any multi-extruder print with a wipe tower.

ex-nerd, 2017-10-31:
I've created a gist containing a cleanup script that seems to work around this issue: https://gist.github.com/ex-nerd/22d0a9796f4f5df7080f9ac5a07a381f

bubnikv, 2017-10-31:
You shall get rid of the automatic priming line if you disable the "single
extruder multi material" checkbox.

ex-nerd, 2017-10-31:
@bubnikv Yes, but that would defeat the purpose of trying to print multiple materials with a single extruder.

bubnikv, 2017-11-06:
Sorry, but as of now the wipe tower in Slic3r Prusa Edition only supports the Prusa Multi Material upgrade. We don't want to be evil or closed source, we just don't have enough resources as of now to whip up a fully parametric version of a wipe tower, and the filament exchange procedure integrated into our wipe tower code is quite complex and it has been fine tuned for our printers. I hope we will improve the wipe tower one day. 

As of now, you may either use the Slic3r PE without the wipe tower, or continue to use your post-processor, or implement the necessary changes into Slic3r and provide us with a pull request.

Relates to https://github.com/prusa3d/Slic3r/issues/559 https://github.com/prusa3d/Slic3r/issues/560

ex-nerd, 2017-11-06:
Thanks for the update, though I specifically wrote this request to just refer to adding a single checkbox (and a handful of accompanying `if` blocks in the code), rather than adding full parametric changes to support every printer variant out there (that's what firmware-based tool change code is supposed to handle -- it "just works" on my printer with `T0` and `T1`).  At the least it would be nice if there was a warning in the preferences (and on the website) that the multi-material features aren't likely to work well on non-Prusa printers.

bubnikv, 2017-11-06:
What exactly is your setup? A diamond hot-end or an E3D cyclop or something similar? 

> Thanks for the update, though I specifically wrote this request to just refer to adding a single checkbox (and a handful of accompanying if blocks in the code), rather than adding full parametric changes to support every printer variant out there

There is more being done behind the scenes. Our Multi-Material setup requires the pulled-out filament to be shaped and cooled down, so at the start of a tool change the active filament is first pushed hard to extrude as much as possible of the hot material from the melt zone. I suppose you would want to disable that as well, right?

ex-nerd, 2017-11-06:
I'm working with a few others to produce an updated spec of the Kossel (Rostock) delta printer with Duet motherboard and [Trianglelab bowden tube splitter](https://www.aliexpress.com/item/3D-printer-E3DV6-Cyclops-dualhead-upgrade-kit-2WAY-in-1WAY-out-TL-Feeder-Prusa-bowden-splitter/32770576556.html), currently running a standard E3D Lite nozzle (though others are running the regular E3D nozzle).  We're actually trying pretty hard to keep this modification as close as possible to the Prusa multi-material upgrade.

The Duet handles the tool-change code in "firmware" via specific [tool change macro gcode files](https://duet3d.com/wiki/Configuring_RepRapFirmware_for_a_Cartesian_printer#Tool_change_files) that get run in sequence whenever a `T#` gcode is received.  In the case of my printer, these are doing pretty much exactly what Prusa Slic3r does in its own currently hard-coded tool change (retract filament A, possibly change temperature, extrude filament B), except the geometry is configured specifically for my printer.  The internal diameter of my splitter seems to limit the need to "shape" the filament, but I'd be doing something similar if it didn't do that.

I guess I'm just assuming Slic3r is doing something similar to this, but with hard-coded pre/post macros in the gcode generator of the slicer engine.  If this is true, it should be simple enough to just wrap that section of code in `if (use_prusa_toolchange_gcode) ...` (or whatever).

bubnikv, 2017-11-07:
Thanks for all the information.

>I guess I'm just assuming Slic3r is doing something similar to this, but with hard-coded pre/post macros in the gcode generator of the slicer engine. If this is true, it should be simple enough to just wrap that section of code in if (use_prusa_toolchange_gcode) ... (or whatever).

There is a virtual class Slic3r::WipeTower defined in GCode/WipeTower.hpp, from which a Slic3r::PrusaMultiMaterial::WipeTowerPrusaMM class is derived. You may just copy the WipeTowerPrusaMM into a WipeTowerGeneric class and remove all the Prusa3D specific stuff from it.

Historically the WipeTowerPrusaMM has been derived from our post processor code, which ran out of the Slic3r and it was supposed to be Slic3r agnostic. There is a lot of hard coded stuff in that code and we would like to clean it up and make it generic one day. We are actively looking around for new hires to extend our Slic3r team now to accelerate improvements like this.

bubnikv, 2018-08-02:
The priming areas are now configurable by the new single_extruder_multi_material_priming configuration value. It will be part of the upcoming 1.41.0-beta.

closed 2018-08-02 by bubnikv