prusa3d/PrusaSlicer #558: [Feature Request] "Move outside print" option: anti-fuzz option for mold printing.

scalablecory, 2017-10-29:
**What**

This option would force the head to circle around the outside of the print when making a travel move, rather than crossing the inside.

**Why**

Imagine a simple mold: a cup, that you will pour into to mold a cylinder. The mold has very thin walls -- an economical 3 shells thick, so no infill. And you want a uniform look, so you specify a random seam position.

Currently, the head will make large moves across the inside of the mold when it switches layers and moves to the next random position. With some filaments, this can mean the inside of your mold having a ton of wispy plastic that needs cleanup to avoid embedding it into the final molded item.

This would be solved by forcing the head to travel around the outside of the object, because in the case of mold making you don't care anything about the outside of the mold: it's only the surface finish of the inside area that needs to be spotless.

bubnikv, 2017-11-06:
Do you have an image or a model of the mold?

You may try to enable the "avoid crossing perimeters" feature, but as of now it does not work very well in Slic3r, see
https://github.com/prusa3d/Slic3r/labels/path%20planning

bubnikv, 2018-02-12:
No response received, closing.

closed 2018-02-12 by bubnikv