prusa3d/PrusaSlicer #560: No way to disable Prusa-specific toolchange movement gcode

ex-nerd, 2017-10-29:
### Version
1.37.2

### Operating system type + version
MacOS 10.13

### Behavior

My printer (Duet board) has its own firmware gcode for managing tool changes, and as such only `T0` and `T1` are needed.  However, Slic3r seems to add a bunch of its own gcode (wrapped inside of some `; CP TOOLCHANGE ...` comments for `UNLOAD` and `LOAD`) for both temperature change and for movement.

* I've documented the issues with the temperature change at #559.
* The movement changes are hard-coded to Prusa printers, despite this version of slic3r being advertised as working fine with other printers.
* This code also makes a number of calls to `M907 E550`, which doesn't exist in the Duet firmware (thankfully it's ignored, but it does pop up tons of warnings throughout my prints).

The movement changes also seem to interfere with my firmware-managed tool change movements (filament is intermittently being double-retracted during tool change, resulting in "empty" layers), though I can't pin down any specifics until I finish testing a cleanup script to remove all of the prusa-specific movements and make sure it's not a bug in my own printer's code.

#### STL/Config (.ZIP) where problem occurs
Any multi-extruder print

ex-nerd, 2017-10-31:
I've created a gist containing a cleanup script that seems to work around this issue: https://gist.github.com/ex-nerd/22d0a9796f4f5df7080f9ac5a07a381f

m0n5t3r, 2017-11-30:
your script is just commenting everything on my setup :)

since I run a regular marlin board, I just rewrote the relevant parts to comment out Prusa's fancy motor current settings, remove the annoying pause and change the length of load / unload extrusions to fit my setup (need a total of 130mm instead of whatever the prusa thing is, I'm too lazy to add them up)

I posted my version at https://gist.github.com/m0n5t3r/66c06090bfc6faf9f4b638382222c97e in case anyone else needs it

bubnikv, 2019-08-01:
The upcoming PrusaSlicer 2.1.0-alpha1 will contain code for wipe tower & multiple extruders. Such configuration will not emit the Prusa Multi-material specific filament exchange sequences. Does it satisfy your needs?

lukasmatena, 2019-10-01:
No response, closing. The wipe tower should now (PrusaSlicer 2.1.0) be configurable enough to turn the "Prusa-specific" things off and/or override it in custom toolchange gcode.

If you believe otherwise, let us know. We don't have resources to do proper testing on all (un)imaginable printer setups that live there in the wild, so we rely on input from our users.

closed 2019-10-01 by lukasmatena