prusa3d/PrusaSlicer #561: change bed temperature at specific layer

panch47, 2017-10-29:
Hi,
I'm using slic3r prusa edition and it is amazing ! Thank you a lot for your work.
Would it be possible to change the bed and extruder temperature at specific layer instead of just the first one?
I'm using a specific filament and I need to start my bed at 70°C. Then I progressively cool it so I manually add lines at different layers to change the bed temp. I set the temp at 60 when Z=2 ; 50 when Z=4 ; 40 when Z=6.
But it is a lot of work every time I generate a gcode file.
If I don't do that, my print bend a lot even with 8 mm of brim.

Thank you !

mikko70, 2017-11-15:
I think it not necessary drop temp linear, but too agressive drop can cause thermal error in printer, about 4 celsius is safe and drop always quite linear.

Bending caused if you filament temp is too high. 70 celsius is too high for PLA
Sometimes i use same with panel, 60- 40 celsius about 5 celsius per drop.
Energy save is good reason for this option.

This option is very easy add in layer view and can give temperature visual feedback if not want do it.

panch47, 2017-11-15:
I don't use 70° for PLA. I use 70°C for a specific filament "bigrep pro HT". And I have tried many thing to prevent bending and decreasing the bed temperature is the best thing I have found. Energy save is also a good thing because the bed consumes a lot of energy to heat.

Where is this option in layer view you are talking about? Thanks

bubnikv, 2017-11-28:
The upcoming build will have conditional expressions, which you may use to fully program the temperatures. In your before_layer_gcode, you may write an expression based on layer_num and / or layer_z. For example

{if layer_z<your_layer_max_z}
M104 S{yourmintemp+(layer_z/your_layer_max_z)*(yourmaxtemp-yourmintemp)}
{endif}

Should do the trick.

panch47, 2017-11-30:
that is great ! thanks a lot for your work, I'll wait for the new build and try it

panch47, 2018-01-04:
wow that's great ! it works perfectly on gcode ! Thanks a lot dude for all your works on this !

rongith, 2018-01-30:
Can you close the issue? It is nice to see issues number decreasing. ;-)

bubnikv, 2018-01-30:
Thanks, closing.

closed 2018-01-30 by bubnikv

panch47, 2018-01-30:
Hi, thank you for closing, I didn't know I had to. The release is working
great for the bed temp changing layer