prusa3d/PrusaSlicer #562: Soluble Interface Support: Reduce distance of infill in prime tower

jsiebert, 2017-10-31:
### Version
1.37.2

### Operating system type + version
Windows 10 Pro 64bit (1709)

### Behavior
When printing in Soluble Interface Support mode (using PrimaSelect PLA + PVA), the hollow part of the prime tower isn't covered with filament after the first filament change occured (from PLA to PVA). The reason might be that the PVA is not able to cross such big distances as they are given with the current design of the infill pattern of the prime tower.
 
The result is that filament strings are dangling around which might get picked up by the extruder resulting in a damaged prime tower. The prime tower gets never closed at all.

[image]

### Possible solution
One solution could be to reduce the distance of the infill structures in the hollow part of the prime tower so that the PVA has the chance to stick better to the PLA underneath.

Another solution could be to close the hollow part of the prime tower using one or two layers of PLA before the first filament change occurs,

gregspruce, 2017-11-01:
I'm experiencing this same issue using support material. I'd really like more settings for the Wipe Tower exposed to the user if possible.

bubnikv, 2017-11-06:
relates to
https://github.com/prusa3d/Slic3r/issues/563
alexrj#3316