prusa3d/PrusaSlicer #563: Soluble Interface Support: Improve eveness and adhesion of Soluble Interface

jsiebert, 2017-10-31:
### Version
1.37.2

### Operating system type + version
Windows 10 Pro 64bit (1709)

### Behavior
When printing in Soluble Interface Support mode (using PrimaSelect PLA + PVA), the Soluble Interface is directly printed onto the support structure created from PLA. The PVA does not stick to support structures very well and sinks into the open spaces between the support structures. Also the PVA does not stick to the support structures very well at the edges of the support.
This results in an uneven surface of the part that should be supported by the support structure. 

Increasing the number of interface layers doesn't help in this case because the parts of the interface which do not stick to the support start to rise resulting in even more unevenness in the surface of the supported part (especially at the edges).

[image]

### Possible solution
Close the support structure using one, two or three layers of PLA to get an even surface for printing the Soluble Interface with PVA. This should also improve the adhesion of the PVA to the support structure.

bubnikv, 2017-11-06:
This is another one on my TODO list for a long time.
Also this has been discussed a long time ago:
https://github.com/alexrj/Slic3r/issues/3316

Locutus84, 2018-11-15:
I am experiencing the same issue with the MMU2 and BVOH, printing on PLA supports with a 2mm spacing.
The first interface layer can be considered as bridging the underlying supports. So it should (like all bridges) only move in straight lines perpendicular to the supports.
Not move in intricate patterns on top of air. Nothing to stick to.

Lukrative525, 2020-03-22:
@Locutus84 this is an old issue but I'm having problems with this and your proposed solution seems like a good one.

@bubnikv can this solution be implemented?

bubnikv, 2021-02-24:
Fixed with 32db22b77c73dd1579da686b748267ce5ee1aeb5

    Fix of Improve Support material adhesion by base type interface layers
    under soluble supports. #5823
    Implemented as a pull request #5903 by @spiky2021
    and reworked.

Also the 1st layer will be printed with rectilinear pattern and its density will  be configurable.

closed 2021-02-24 by bubnikv