prusa3d/PrusaSlicer #564: version of Slic3r Prusa edition or PrusaControl running on arm H3 (armhf or appimage) ?

openelearning, 2017-11-01:
Hello,

I'm trying to make possible to deploy open source and open hardware nano computer and 3D printing for schools in France. Actually it seems that Original Prusa i3 MK3 and H3 SBC are a good choice for this. In order that the children could slice their 3D object, a slicer is needed, but it seems that actually there isn't Slic3r Prusa edition or PrusaControl running on arm H3 (armhf or appimage).

### Operating system type + version
I'm using armbian for Orange PiPC+, the ubuntu xenial 16.04 legacy kernel that run armhf https://www.armbian.com/orange-pi-pc-plus/

### Behavior
* _Describe the problem_
I don't know how a children using stl files could make gcode for Original Prusa i3mk2 /mk3 with an H3 arm Single Board Computer.
* _Expected Results_
The children could have a nice gcode that runs on Prusa i3 MK2/3 :)

#### Other information that could help
* OBS can now make AppImage that are running on H3 SOC - armhf, as t-paul has greatly made an OpenScad version that is running on H3 boards : https://github.com/openscad/openscad/issues/1849#issuecomment-340319476

* It seems that someone has used a part of slic3r on RBPI : https://github.com/prusa3d/Slic3r/issues/506

* https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi 

* https://github.com/OctoPrint/OctoPrint-Slic3r

* http://bukobot.com/raspberry-pi-slice

Is it possible to have a version of Slic3r Prusa edition or PrusaControl running on arm H3 (armhf or appimage) ?

If yes, what I can do to help for this ? (I'm newbie in building package / compilation / using OBS, but I can spend time on learning, searching information, try to make crowdfunding for this...)

Thanks for your attention and help

openelearning, 2017-11-02:
There is also already Slic3r projects in OBS : enter slic3r in https://build.opensuse.org/search

bubnikv, 2017-11-02:
Will the old Slic3r 1.2.9 run for you?
https://packages.debian.org/search?keywords=slic3r&searchon=names&suite=stable&section=all

I may get in touch with Chow Loong Jin, who maintains the Debian package, to get a hint why / how the Slic3r PE is failing to build on ARMs at the Debian's build servers.

Frankly porting Slic3r to ARM does not have much of a priority as the slicing is too slow on those little critters.

openelearning, 2017-11-02:
Thanks a lot for your reply.
The old Slic3r 1.2.9 seems not running for me on Orange Pic PC+ with armbian ubuntu desktop legacy : 
I have installed it with sudo apt-get install slic3r, and here what I have if try to launch it with terminal :
```
~$ slic3r --gui
Can't locate Wx.pm in @INC (you may need to install the Wx module) (@INC contains: /usr/bin/lib /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.22.1 /usr/local/share/perl/5.22.1 /usr/lib/arm-linux-gnueabihf/perl5/5.22 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.22 /usr/share/perl/5.22 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base .) at /usr/share/perl5/Slic3r/GUI/AboutDialog.pm line 6.
BEGIN failed--compilation aborted at /usr/share/perl5/Slic3r/GUI/AboutDialog.pm line 6.
Compilation failed in require at /usr/share/perl5/Slic3r/GUI.pm line 8.
BEGIN failed--compilation aborted at /usr/share/perl5/Slic3r/GUI.pm line 8.
Compilation failed in require at (eval 65) line 1.
```

I can't have the user interface, slic3r doesn't appear.

bubnikv, 2017-11-06:
Sorry, we don't have resources as of now to prepare an ARM build. You may try to compile the Slic3r PE on your ARM and I will try to help you to solve your issues along the way. But it may take many iterations and I cannot promise a success.

We don't have an up to date build process description for LInux as of now, but the following should mostly work:
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux

you will also need to install the intel thread building block package.

openelearning, 2017-11-20:
Thanks a lot for your reply, sorry for the delay I was searching partners and help for the project and trying to make a SBC x86 demo in order to start making demonstration.

Thanks a lot for your help !
Before trying to compile Slic3r PE on my ARM, do you thinks using OBS (http://openbuildservice.org/help/) could be more easier for a newbie as me ? 

I have tried to to make this : https://build.opensuse.org/package/show/home:Ucino/Slic3rPrusaEdition but it doesn't work actually.

I have looked the spec files of other user that have build slic3r standard (https://build.opensuse.org/search and enter slic3r) (ie: https://build.opensuse.org/package/show/home:cgiboudeaux/slic3r ).

Do you think it will be easier than building on the arm ? (if yes I will try to solve the obs problem, if no I will try to learn to compile it on the ARM).

bubnikv, 2017-11-28:
@openelearning Sorry, I cannot answer any of your questions without investing a lot of time.

openelearning, 2017-11-28:
@bubnikv  Ok, thanks for your reply, I will try to learn OBS, if I failed, I will try to to learn to compile the Slic3r PE on my ARM.

openelearning, 2017-11-30:
thanks to @eyal0 that give these informations : "These are the instructions that I used for compiling Slic3r on a RaspberryPi with OctoPi: https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi"

openelearning, 2017-11-30:
On https://packages.debian.org/sid/slic3r-prusa we can find that there is an armf version of slic3r PE that's a very good news :) 
However on  the armbian for orange PiPC+ ubuntu xenial 16.04 legacy, we can not run them with gdeby because of perl : it's 5.22, and this version of slic3r 1.38.0+dfsg-1 on sid need 	perl >= 5.26.1-2
So it seems that appimage is the easiest way to make slic3r PE running on orange pi H3 SBC on armbian. So OBS seems to be a nice to try, as OBS can make appimage for arm.

I will try to learn how I can make a nice spec file https://build.opensuse.org/package/show/home:Ucino/Slic3rPrusaEdition. It seems that one of the step for this is to complete the dependencies for the build on the spec file, i think that https://packages.debian.org/sid/slic3r-prusa and  https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi can help me for that. Let's see...

eyal0, 2017-11-30:
@openelearning I tried to install the armhf package on my OctoPi Raspberry Pi but it didn't work.  I think that it's due to missing dependencies.

```
The following packages have unmet dependencies:
 slic3r-prusa : Depends: perl (>= 5.26.1-2) but 5.20.2-3+deb8u9 is installed
                Depends: perlapi-5.26.1 but it is not installable
                Depends: libboost-atomic1.62.0 but it is not installable
                Depends: libboost-chrono1.62.0 but it is not installable
                Depends: libboost-date-time1.62.0 but it is not installable
                Depends: libboost-filesystem1.62.0 but it is not installable
                Depends: libboost-locale1.62.0 but it is not installable
                Depends: libboost-log1.62.0 but it is not installable
                Depends: libboost-regex1.62.0 but it is not installable
                Depends: libboost-system1.62.0 but it is not installable
                Depends: libboost-thread1.62.0 but it is not installable
                Depends: libglew2.0 (>= 1.12.0) but it is not installable
                Depends: libstdc++6 (>= 5.2) but 4.9.2-10 is installed
                Depends: libtbb2 but it is not installable
                Depends: libwxbase3.0-0v5 (>= 3.0.3.1+dfsg2) but it is not installable
                Depends: libwxgtk3.0-0v5 (>= 3.0.3.1+dfsg2) but it is not installable
                Recommends: libopengl-perl but it is not installed
                Recommends: libwx-glcanvas-perl but it is not installed
                Recommends: libxml-sax-expatxs-perl but it is not installed
                Recommends: libpdf-api2-perl but it is not installed
                Recommends: libio-all-perl but it is not installed
                Recommends: libsvg-perl but it is not installed

```

All the libboost 1.62 stuff is unavailable.  I also had this problem when trying to compile >=1.38.0.  I can't find that version from `sudo apt-get install` on my Raspberry Pi.  What repository should I be using for Raspberry Pi3 jessie?

Thanks.

bubnikv, 2017-12-05:
@eyal0  I wanted to accept your legacy boost compatibility patch, but I cannot find it. Would you please remind me where did you post it? Thanks.

eyal0, 2017-12-05:
@bubnikv I intentionally didn't make a PR because I think that supporting old boost is a bad idea.  I was able to install new boost on RPi and compile the code without these changes.  And the changes make the code look worse, not better.  For most people, having a modern libboost won't be a problem.  So I think that you should not do these changes.  And you're not running old boost so you might at any point re-break old boost compatibility and we'd have to make another patch.

That said, here they are: https://github.com/eyal0/Slic3r/commit/af6eae7317a4c0db13fd80c9088bc2a60d7f9198

I also need to use this link to get it working: http://www.boost.org/doc/libs/1_65_1/boost/spirit/home/qi/copy.hpp

I didn't copy the first line of that function, not sure if it'll work.  Actually, since I've already installed the new libboost on my system, I no longer have an easy way to test if it's broken with Boost1.55 unless I reformat my RPi!

bubnikv, 2017-12-05:
> That said, here they are: eyal0/Slic3r@af6eae7

Thanks, I will have to sleep on it.

openelearning, 2017-12-06:
@hyperair maybe you have some ideas where I can try to investigate in order to have https://build.opensuse.org/package/show/home:Ucino/Slic3rPrusaEdition# working for appimage armv6l or armv7l (they are both excluded), for example if I had to find some package to add in BuildRequires: in https://build.opensuse.org/package/view_file/home:Ucino/Slic3rPrusaEdition/slic3rPrusaEdition.spec?expand=1 ?

openelearning, 2017-12-06:
This page will be very usefull : https://github.com/AppImage/AppImageKit/wiki/Using-Open-Build-Service & https://build.opensuse.org/package/show/OBS:AppImage/AppImageKit

openelearning, 2017-12-07:
Here some new test using OBS to make armv7 appimage (without success actually, I have 2 errors : nothing provides perl(OpenGL), nothing provides perl(Wx::GLCanvas) ) : https://build.opensuse.org/package/show/home:Ucino:branches:OBS:AppImage:Templates/Slic3rPrusaEdition#

bubnikv, 2017-12-07:
> branches:OBS:AppImage:Templates/Slic3rPrusaEdition#

You may need to build and install the two packages using cpan. Those
packages would likely be installed by running
perl Build.PL
from inside the Slic3r directory.

But take these advices with a grain of salt, I have no experience with the
OBS, so I don't know what kind of functionality it offers.

openelearning, 2017-12-07:
Thanks a lot for your help @bubnikv  :) luc14n0 is helping to make an appimage on https://build.opensuse.org/package/show/home:luc14n0/slic3r and he solve the problem of perl(OpenGL), and perl(Wx::GLCanvas). He's trying to solve other problems now.

bubnikv, 2019-04-12:
You may guys try to compile the current Slic3r master on ARM.
https://github.com/prusa3d/Slic3r/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

A command-line slic3r is compiled with an additional cmake switch 

`-DSLIC3R_GUI=no`

bubnikv, 2019-05-24:
No response. I was able to compile PrusaSlicer on an Android e-ink reader, so you should be able to compile on whatever reasonable ARM distribution you pick.
Successful headless compilation was reported on R-Pi.
Closing.

closed 2019-05-24 by bubnikv

openelearning, 2019-05-24:
I'm really sorry for not having replied before. Thanks a lot for the link, this will help me, and this really usefull that we can now use PrusaSlicer on arm distribution. For the moment I have give up the use nanocomputer to make gcode, but I will try later, and give you feedback. The project is going well, French schools want to benefit it, and the project has been nominated for an innovation competition. A pedagogical project have been also made to show the interest of open source and open hardware solutions :  https://www.open-elearning.com/magano/magano-education/monumentuss/

openelearning, 2019-05-24:
(for the moment nanocomputer has been replaced by old computer as you can see on this project : https://www.open-elearning.com/magano/magano-education/equiper-ethiquement-les-ecoles-primaires/ the goal of this second project is to find solutions to make easy to any school to have open source and open hardware digital tools (computer, 3D printer, IA module...).

probonopd, 2019-05-24:
@bubnikv shall we make an AppImage for Raspberry Pi?

bubnikv, 2019-06-03:
@probonopd appimage for Raspberry PI would be nice. We need to find some
time to set up a build server for another target.

probonopd, 2019-06-03:
We produce some ARM binaries for https://github.com/AppImage/AppImageKit/releases using Travis CI and Docker images for crosscompiling. Maybe @theassassin can give you some hints.

bubnikv, 2019-06-03:
Thanks for the hint.

mmiscool, 2020-05-11:
> No response. I was able to compile PrusaSlicer on an Android e-ink reader, so you should be able to compile on whatever reasonable ARM distribution you pick.
> Successful headless compilation was reported on R-Pi.
> Closing.

Any details on how to compile and run on android for the command line version. I have managed to use busy box on android and would love to use this thing.

bubnikv, 2020-06-11:
Got it working, but it is quite convoluted and the software 3D rendering is very slow.
https://twitter.com/bubnikv/status/1270966793419526145