prusa3d/PrusaSlicer #565: Wierd multiple printer entries in the controller window

quirxi, 2017-11-02:
### Version
Slic3r-1.37.1-prusa3d-linux64-full-201709141215

### Operating system type + version
$ uname -a
Linux debian 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u5 (2017-09-19) x86_64 GNU/Linux
$ cat /etc/debian_version 
9.2

### Behavior
* Slic3r's Controller Window shows multiple entries for the same device. It seems as if the entries were on a per printer basis but one printer shows up twice.

* _Steps needed to reproduce the problem:
1. Open slic3r
2.  Load the attached configuration file
4. Have a look at the controller window.

* Expected Results: Since one device can normally only have one printer attached to it, it would make sense to group the entries per device and let the user choose which printer should be related to this device. Also double entries of a printer (and even for one device!?) should not be allowed.
* Actual Results: A device can have multiple printers attached to it and one printer even shows up twice.

[Slic3rControllerWindowError.zip](https://github.com/prusa3d/Slic3r/files/1437274/Slic3rControllerWindowError.zip)

bubnikv, 2017-11-06:
Would you please export and attach a config bundle file?
Thanks.

quirxi, 2017-11-06:
Its funny because now it had less entries but there are still the double entries that I described above. 

[Slic3r_config_bundle.ini.zip](https://github.com/prusa3d/Slic3r/files/1447279/Slic3r_config_bundle.ini.zip)

bubnikv, 2020-03-20:
Controller is long gone, we do not currently support printing over serial line / USB in PrusaSlicer.
Closing.
If we ever impelment printing over USB  / serial line, we will have a new batch of bugs.

closed 2020-03-20 by bubnikv