prusa3d/PrusaSlicer #566: Extrusion width for first layer too small?

m-reiter, 2017-11-02:
### Version
_1.37.2-prusa3d_

### Operating system type + version
_OS X 10.11.16_

### Behavior
* I'm not 100% sure whether this is an actual issue with Slic3r or a misunderstanding on my part. I see an EW for the first layer in GCode comments created by Slic3r which doesn't seem to fit my settings.
* _Steps needed to reproduce the problem_
  * start with default settings
  * set layer height to 0.1mm, first layer height to 0.4mm
  * slice any STL file (I would attach a sample but I keep getting "Something went really wrong" errors)
* _Expected Results_
  * first layer extrusion width should be 0.8mm (200% of first layer height)
* _Actual Results_
  * as per the GCode comments, first layer EW is 0.2mm

Sebastianv650, 2017-11-05:
Do you mix up extrusion width with height? Can you add a gcode file as an example for what happens please.

m-reiter, 2017-11-06:
I don't think I do, or at least if I do, Slic3r does as well. I've attaches a gcode file that shows the issue. Note that in this case, other settings were changed as well. I can however reliably reproduce this with changing just the layer heights from the defaults.

This is from the file's header:

`; generated by Slic3r 1.37.2-prusa3d on 2017-10-29 at 16:36:27

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.20mm
`

I haven't done any reverse calculations from the gcode but clearly Slic3r thinks first layer extrusion width is 0.20mm.

[heattower.zip](https://github.com/prusa3d/Slic3r/files/1445472/heattower.zip)

bubnikv, 2017-11-06:
You specified the first layer width to be 200% of the first layer height, that should be 0.8mm.
You get 0.2mm, which is 200% of the layer height, not the first layer height. So there seem to be a regression bug in Slic3r PE when specifying the first layer width in percents.

Unfortunately this happens during the porting from Perl to C++.

bubnikv, 2017-11-06:
Actually it was a regression bug in the G-code comment only. I have fixed it with https://github.com/prusa3d/Slic3r/commit/71b58e24a9671f6a6d9f1e0703505090fd755266
Thanks for the report.

closed 2017-11-06 by bubnikv

m-reiter, 2017-11-06:
Thanks for the quick reaction! I had a suspicion it might be the comment only - probably would have been noticed sooner otherwise.