prusa3d/PrusaSlicer #568: Custom Filament G-code for all filaments is being added to exported G-code regardless of which filament is selected

liftoff, 2017-11-03:
### Version
1.37.2 (latest Prusa edition as of today)

### Operating system type + version
Ubuntu 17.04

### Behavior
I added "custom G-code" to a number of filaments (e.g. `M203 E15` for the starting g-code and `M203 E120` for the ending g-code for Ninjaflex) and when Slic3r generates g-code it puts the start and end filament custom g-code values for *all* filaments at once.  So at the start of the print it'll put the filament g-code for tools 1, 2, 3, and 4 in a row which completely screws up the point of that feature.  The gcode for each individual filament should be set when a tool change occurs or at the start of the print.

#### STL/Config (.ZIP) where problem occurs
Here's an example of the g-code output by Slic3r:

    G92 E0.0
    ; Filament gcode
    M203 E30 ; Max extrusion speed 30mm/sec
    ; Filament gcode
    M900 K125
    M203 E15
    ; Filament gcode
    M900 K40
    ; Filament gcode
    M900 K40
    G21 ; set units to millimeters
    G90 ; use absolute coordinates
    M83 ; use relative distances for extrusion
    T1
    ;BEFORE_LAYER_CHANGE
    ;0.2

In that print I was *just* printing Ninjaflex (no other tools/filaments) so what ended up happening was the linear advance K value was incorrectly set to `K40` when it should've been `K125`.  Even though the filaments in tools 3 and 4 were not used in the print their start and end g-code settings were included.  This should not be happening.

What *should* happen is the start g-code should be set after a tool change occurs and only for the filament used by that tool/extruder.

bubnikv, 2017-11-06:
I have had a similar internal request recently. I may either add a "filament exchange" g-code section, or emit the filament g-code at the tool change for the "single extruder multiple material" printers, or both.

m0n5t3r, 2017-11-23:
since you generally only use one tool at a time (from what I've seen, mixing extruders also define virtual tools for a specific mix), I think it should be safe to emit the filament specific gcode on tool change in all cases

the current behaviour is pretty annoying, especially when using flexible and rigid materials in the same print (vastly different linear advance settings, as seen above)

bubnikv, 2017-11-30:
I have implemented your request with https://github.com/prusa3d/Slic3r/commit/3996535e5dea9d223fe5ad9e628bb3b841aeff63 and it will be part of the next alpha build, so please provide a feedback in case it will not work as expected.

The commit comment follows:

G-codes in case of single extruder multiple material setup:
At the start of the print, the filament_gcode_start is executed
for the active extruder only, and the filament_gcode_start /
filament_gcode_end are then executed at each tool change.

When the Prusa MM wipe tower is active, the tool changes are handled
a bit differently: M900 K0 is emited before the wipe tower extrusions start,
and the filament_gcode_start code is executed after the wipe tower extrusions
are done. This rule effectively disables the linear advance over the wipe tower.

closed 2017-11-30 by bubnikv

liftoff, 2017-11-30:
Awesome!  I look forward to testing this.

m0n5t3r, 2017-12-07:
Doesn't seem to work with single material prints:

* with "single extruder multi material" unchecked, it outputs both filament start gcodes (old behaviour)
* with "single extruder multi material" checked and a single extruder being used (either 1 or 2, in my case) outputs no filament start gcode at all
* with "single extruder multi material" checked and a multi-material part it behaves correctly

Slic3r Prusa Edition
Version: 1.38.3-prusa3d-linux64
Build: Slic3r-1.38.3-prusa3d-linux64-full-201712012144
Operating System: linux
System Architecture: x86_64-linux-thread-multi