prusa3d/PrusaSlicer #570: 0.25mm nozzle support for wipe tower

robotrovsky, 2017-11-04:
Hi!

I know you guys are busy, but can you tell me if support for the 0.25mm nozzle in combination with the wipe tower aka mmu is planned?

Thanks

robotrovsky, 2017-11-05:
I understand that this will not be possible. Someone in the forum explained me that there will be to few extruder steps for the 0.25mm nozzle.

Sebastianv650, 2017-11-05:
Can you provide the link to this information? I would also like to know why this should be impossible.

robotrovsky, 2017-11-05:
Of course:
https://shop.prusa3d.com/forum/original-prusa-i3-mk2-multi-material-f26/0-25mm-nozzle-and-multi-material-t6509.html

Sebastianv650, 2017-11-05:
OK I see the logic behind. But that's only a reason for that specific extruder, not for Slic3r as a generic tool in general. So if Slic3r doesn't allow you to activate the wipe tower at the moment for specific nozzle sizes, I would keep this issue open.

Only my two cents: I would grab a 0.25mm nozzle and try it on my own if I where you. Maybe the result is OK for your prints, and the cost for an extra nozzle is low. In worst case, you learned something about steady extrusion flow and esteps ;)

robotrovsky, 2017-11-05:
I don't want to bother the guys with physically challenging (avoiding the word impossible) issues ;-) I already ordered 0.25mm nozzles, so i will try anyways, but even if they work, the really interesting thing would have been to use it with soluble support. I really like the possibility of multiple colors, but over the time i see the disadvantages coming with the bowden extruders :-/

Sebastianv650, 2017-11-05:
If the 0.25mm nozzle works in your bowden system for low layer heights it will also work for MM printing. The steps/mm isn't changed between using one material or 4. And as said before, Slic3r is not limited to 140steps/mm printers. So there is not reason to block this feature in general.

> I really like the possibility of multiple colors, but over the time i see the disadvantages coming with the bowden extruders :-/

I know - that's the only thing holding me back from upgrading. It would have been awsome if they would have kept the direct drive, only using the 4 aditional steppers for the filament change itself and unclutching them when done.. I hope they go this way, maybe for a version 2.0.

robotrovsky, 2017-11-05:
> If the 0.25mm nozzle works in your bowden system for low layer heights it will also work for MM printing. The steps/mm isn't changed between using one material or 4.

Yes, that is obvious. But again, the Wipe Tower doesn't support the 0.25mm nozzle, so in theory it would be possible, but not practically as long as i am not willing to handwrite gcode for the wipe-tower or an post processor. This week i have no access to my printer, but the 0.25mm nozzle isn't even listed in the SingleMaterial profiles. I guess the guys at prusa are so smart to adjust some ini-files and have tried it, that the profile is not there makes me pretty sure, that it doesn't work at all. Also the Wipe Tower doesn't even support variable layer height. I can't print half the object in 0.15mm and the other half in 0.20mm although both layer heights are supported. So i think there is a lot of work to do improving the wipe tower, with some easier to achieve goals, that might/should be implemented before.

robotrovsky, 2017-11-05:
And if we dream let them also integrate a wipe-bucket and a wipe-mechanism to get rid of the annoying wipe tower and all the problems and garbage plastic it produces ;-)

Sebastianv650, 2017-11-05:
> That is a nice idea, i would completely support. Its just one motor more...

A bit more might be necessary. Each stepper must be able to clutch in and out, which means something like a servo has to separate the extruder drive gears from the filament so it can be moved by the other stepper. Definitly possible, there are designs for such switching extruders but it's another problem to solve beside the shaping of the half-melted filament tip. I guess therefore they choosed the "simplified" design with bowden drives.

> And if we dream let them also integrate a wipe-bucket and a wipe-mechanism to get rid of the annoying wipe tower and all the problems and garbage plastic it produces ;-)

Even if a bucket is used instead of a tower, it needs to spit out the same amount of filament during the change. Only the perimeter and 5% (?) infill on a layer without color change could be safed. Not worth the extra work I think.

gregspruce, 2017-11-06:
> Even if a bucket is used instead of a tower, it needs to spit out the same amount of filament during the change. Only the perimeter and 5% (?) infill on a layer without color change could be safed. Not worth the extra work I think.

True, but then you could be specific for each filament swap regarding exactly how much needs to be purged until it is sufficiently clean. Ie. Going from Fillamentum Nobble Blue to nearly any color besides black needs like 4 times as much purge as changing from Rapunzel Silver to anything else.

And that way, no logic would be needed layer-to-layer to ensure a footprint for higher layer purges :).

The more I think about it, the more I really want a purge bucket option.

bubnikv, 2017-11-06:
I just checked with Jindrich, who is responsible for fine tuning and testing the Slic3r and Prusa Control profiles. The 0.25mm nozzle did not work with MM, it jammed too often.

The whole MM is quite involved, though it may not seem to as long as you are using our profiles with our MM printers and it works. There is a devil in the detail of the filament exchange sequence, which has to be fine tuned for the particular nozzle geometry and a filament material. We did not have time yet to do the fine tuning for a 0.25mm nozzle and it may not even be possible to make our current method work with the current extruder and a stock 0.25mm nozzle, as the tiny nozzle orifice may not allow us to ram the hot material from the melt zone quickly enough.

> It would have been awesome if they would have kept the direct drive, only using the 4 aditional steppers for the filament change itself and unclutching them when done.. I hope they go this way, maybe for a version 2.0.

Maybe, but there is a lot of detail to be solved here, namely you must not deform the pulled out warm filament with the extruder hobbled bolt, otherwise you will not be able to re-insert it.

> And if we dream let them also integrate a wipe-bucket and a wipe-mechanism to get rid of the annoying wipe tower and all the problems and garbage plastic it produces ;-)

The bucket does not really help. You need a lot more space for the bucket than for a compressed wipe tower. And the purge works better if it has the back-pressure from the underlying purge tower layer.

robotrovsky, 2017-11-06:
> The bucket does not really help. You need a lot more space for the bucket than for a compressed wipe tower. And the purge works better if it has the back-pressure from the underlying purge tower layer.

That must not be, maybe there is no bucket just a slide to the side of the heating bed. That was what i was thinking about with "redirecting to the ground". Also the wipe tower has more problems:
https://github.com/prusa3d/Slic3r/pull/433
And a failure in the wipe tower can destroy your whole print, as a wipe bucket would not be relying on a perfect process. Also it is easier to make the extrusion settings per filament etc. Just my opinion.

bubnikv, 2017-11-06:
Do you want to install a wipe bucket on the MK2 printer or do you use any
other printer? Do you have an idea how and where to install the bucket?

robotrovsky, 2017-11-06:
I have the MK2 and this is just an idea right now. If I say bucket I just think of a little metallic slide with slippery finish and very sharp angle at big height in a dedicated corner of the print bed that makes the extruded filament fall on the ground. There you can put a bucket on the ground next to the printer or make everything dirty =) then you put a little lid and a brush next to the top of the slide to clean the nozzle after extruding. So when changing the filament you move the nozzle up to approximate 20cm drop everything on the slide move to the side over the lid and the brush to clean it. Then you move down again and resume printing. This are just ideas, it is my first printer but I know gcode and the system (steppers etc) from my stepcraft cnc.

robotrovsky, 2017-11-06:
To ensure the nozzle really hits the top of the slide with the lid/brush there could be a pinda probing point on the top of the slide. Just ideas...

bubnikv, 2017-11-07:
> That must not be, maybe there is no bucket, instead just a slide to the side of the heating bed. That was what i was thinking about with "redirecting to the ground".

It sounds easy, but it will likely be quite difficult to control, and the non-squished filament will take a large volume. I have seen wipe buckets at the Sigma 3D printer, but they really have to purge very little, so it is apples vs.  oranges.

Anyway, there seem to be issues with the wipe tower, which we will try to address, for example the stability of a narrow high wipe tower. One simple idea is to rotate the wipe tower by 90 degrees on the bed if it is very narrow, so it is oriented in the direction of the print bed movement.

gregspruce, 2017-11-07:
> Anyway, there seem to be issues with the wipe tower, which we will try to address, for example the stability of a narrow high wipe tower. One simple idea is to rotate the wipe tower by 90 degrees on the bed if it is very narrow, so it is oriented in the direction of the print bed movement.

Please check out KISSlicer's implementation. It does an "L" shape. It might take up marginally more bed space, but it also works well with allowing different purge values for each extruder- which would address a couple other open feature requests and issues,

bubnikv, 2017-11-07:
> Please check out KISSlicer's implementation. It does an "L" shape.

Would you please post a picture? Thanks.

gregspruce, 2017-11-07:
This thread on Prusa forums should cover it, especially the 5th post

https://shop.prusa3d.com/forum/hall-of-fame-best-3d-prints-f15/variable-layer-two-model-4c-mm-print-t6374.html

robotrovsky, 2017-11-07:
> It sounds easy, but it will likely be quite difficult to control,

Of course. This have just been some immature ideas, serving maximum as inspiration ;-)

bubnikv, 2018-08-02:
We have removed the hard coded limits from the wipe tower generator in 1.40 release.
It is now up to you to test it and make it work :-)
Closing.

closed 2018-08-02 by bubnikv