prusa3d/PrusaSlicer #572: "Load Generic..." object modifiers always show "0" as their X/Y/Z position

riskable, 2017-11-10:
### Version
1.37.2

### Operating system type + version
Ubuntu 17.04

### Behavior
When adding a modifier via "Load Generic..." to an object and the moving it around with the sliders the numerical locations always show "0".  Even if you enter a value manually, closing and re-opening the object settings will have their locations reset to 0.  The modifiers still retain their locations (in the print) it's just that the numerical values and the sliders won't reflect their previously-set positions.  Any values changed after re-opening the object settings will be treated as moves from the current position.

In this screenshot I've selected a "lambda-cylinder" modifier where I've previously set the Z to +1 then closed and re-opened the object settings.  You can see that the Z slider has been reset to 0 but it *should* reflect the current position:

[image]

cerdeira, 2017-11-10:
I referenced that on #346, toghether with other things.

bubnikv, 2018-12-11:
I believe we implemented your request in a way you will like. It will be part of Slic3r PE 1.42 alpha, which is to be released soon. Closing.

closed 2018-12-11 by bubnikv