prusa3d/PrusaSlicer #574: STL import error

nikolajkiel, 2017-11-13:
### Version
1.2.9a
https://dl.slic3r.org/win/slic3r-mswin-x64-1-2-9a-stable.zip

### Operating system type + version
Windows 7 64-bit

### Behavior
* Error when importing a *.STL. The geometry is filled with unwanted holes. (The STL file is rendering perfect in MeshLab, SpaceClaim)
* The problem occurs when the included .STL file is imported. (See attached screenshot)

### Attached files
[STL import error.zip](https://github.com/prusa3d/Slic3r/files/1466499/STL.import.error.zip)
[image]

BirkBinnard, 2017-11-14:
That is an interesting STL file. It imports and slices fine in Craftware. Plic3r (Pruse Ver. 1.37.2) makes a mess of it, but also says it has auto-repaired 2274 errors. 

Using the File/Repair STL file ooption results in an OBJ file that still looks bad when loaded. But the message box says the file is Manifold and auto-repaired 1350 errors.

I ran the original STL file through the Windows 3D Builder program which said it found errors in the file. After correcting the file 3D Builder saved it as a much larger file size, but that loaded OK into Plic3r.

This suggests to me that the file has some obscure errors in it that need to be fixed before slicing.

bubnikv, 2017-11-14:
It is one of the "model import" issues https://github.com/prusa3d/Slic3r/labels/model%20import
Slic3r tries to fix the model in 3D, while Cura and likely the other slicers try to fix the model slice by slice, and this strategy seems to be more successful. I will have to look into the model import one day.

enricoturri1966, 2018-01-18:
Fixed with https://github.com/prusa3d/Slic3r/commit/d34f3ffc8320c6929db53233cd2e17d7dac11d01

closed 2018-01-18 by enricoturri1966

FidelCapo, 2018-05-11:
Tested and fixed in Slic3rPE-1.40.0-alpha that will be released soon.