prusa3d/PrusaSlicer #575: Wipe tower causes custom tool change gcode to be ignored

pauln, 2017-11-15:
### Version
1.37.2

### Operating system type + version
Linux (Ubuntu 17.10)

### Behavior
Following on from #557 and #560, enabling the wipe tower seems to prevent custom tool change gcode from being used.  Tool change retraction settings are also ignored.  This further locks out anyone not using Prusa MM, unless their setup requires precisely the same retraction as Prusa MM.  For example,  I'm running a Kossel with splitter (similar to @ex-nerd's setup described in #557), which needs ~140mm of retraction - but with the wipe tower enabled, only about 92mm of retraction happens regardless of tool change retraction or custom tool change gcode, so the second filament can't be extruded.

I'm aware (from discussion in #557) that the wipe tower is currently very Prusa-specific, but I wanted to raise this issue so that it can (hopefully) be kept in mind when a more generic wipe tower is implemented (whenever that may be).

ex-nerd, 2017-11-15:
Out of curiosity, does my post-processing script work for you?  I haven't tested it very much with 1.37.2.

bubnikv, 2017-11-28:
You already mentioned that your newly raised issue overlaps with the issues already raised, so I am closing this issue.

We have a new hire on board, who is now taking over the wipe tower code. He needs a bit time to get up to speed, but there is a good chance that we will make the wipe tower code fully parametric.

closed 2017-11-28 by bubnikv

Patrik-svobodik, 2018-12-18:
@bubnikv is someone still working on it? I am planning to build a mmu2 for my Duet Wifi powered HEVO and I want to handle all these changes in firmware