prusa3d/PrusaSlicer #576: Wrong result when importing multiple STL into one multi-part

henrikkoch, 2017-11-15:
When importing ex. 2 STL files and accept them to be an multiple object the files can be moved around like one.  When entering the object and then selecting extruder 1 for one STL and extruder 2 for the other 2 and then slicing this - it looks very well and correct in the preview - two different colors.

But when printing it only uses the first extruder for both colors?  Why this when it look correct in the preview. Is it not allowed to handle them as and multi object. I need this otherwise the one STL part faces to the heat place - and that needs to be elevated from meshmixer where i did the two STL files.

I belived that slicer has an export fault here

henrikkoch, 2017-11-15:
These are the two STL files
[spire 3D files.zip](https://github.com/prusa3d/Slic3r/files/1476530/spire.3D.files.zip)

Here you can the the sliced result. It even though shows two different filamens on the priming zones.
But it only uses filament 1 for both ???
[image]

henrikkoch, 2017-11-15:
#

bolerro, 2017-11-21:
Looking at the image, it appears you are using the same filament for both Ext1 and Ext2. Having the same name for both appears this could be the cause even though they may be set up as individual filaments. I would give the second filament a different name by adding 'Yellow' to the end of it or whatever you decide. I believe Slic3r is looking for the filament name in the ini file and when it finds the first one, it isn't looking for the second one any longer.

bubnikv, 2017-11-28:
This is a basic scenario, which has been tested many times, so it should work. Please attach your generated G-code, so I can inspect it.

bubnikv, 2018-02-12:
No response received, closing.

closed 2018-02-12 by bubnikv