prusa3d/PrusaSlicer #577: Gap in bottom and top layer of objects when using concentric fill pattern

waynix, 2017-11-18:
### Version
Slic3r-1.37.2-prusa3d-linux64

### Operating system type + version
Debian 9.1

### Behavior
* My top and bottom layerhave a groove in the middle of top and bottom layer of the object I want to print. 
* Download [staubsauger.zip](https://github.com/prusa3d/Slic3r/files/1484473/staubsauger.zip)
* Printersettings: https://github.com/prusa3d/Slic3r-settings/tree/master/Slic3r%20settings%20MK2
* I expect a uniform bottom Surface [image]
* _Actual Results_
  * [image]

#### STL/Config (.ZIP) where problem occurs
[staubsauger.zip](https://github.com/prusa3d/Slic3r/files/1484473/staubsauger.zip)

bubnikv, 2017-11-28:
Thanks for the report. Currently the concentric infill does not apply the gap fill algorithm.

men8ifr, 2022-02-06:
I've come across this problem many times with prusaslicer, not just top and bottom layers bot all using concentric fill. very annoying as concentric would be much quicker and smoother than rectilinear but you can't us it.

men8ifr, 2022-02-06:
I shoudl say my issue is windows not linux.. (V2.3)

vBul, 2022-02-18:
Hello,

I'm longing for the gap fill feature on 100% concentric infill.

I have rotationally symmetrical and functional parts, which need to be 100% for strength.
Concentric infill would save a lot of printing time over Rectilinear, if it was solid:
[image]
Adding more perimeters won't help. It is printed slowly, and also eliminates all bridges...

Please add gap fill to the circular infill algorithm!
thx

bubnikv, 2022-06-29:
Implemented in PrusaSlicer 2.5.0-alpha2. 

In PrusaSlicer 2.5.0-alpha2 we have ported the Arachne perimeter generator from Cura. Thanks Ultimaker and Cura team for keeping Cura open and for their dedication.
With Arachne enabled for perimeters, the Arachne is newly used to fill in concentric infill completely.

Closing.

closed 2022-06-29 by bubnikv