prusa3d/PrusaSlicer #578: Tagging single object layer in multipart print job

bwright86, 2017-11-18:
When printing multiple separate parts in a single job, on occasion one object may start to fail, and would need to be removed from the print job. I would like to see an option in Slic3r that adds a comment tag before and after each object's layer in the print.

This idea comes from single part prints when an issue arises (i.e. axis offset after bump, power-loss, multi-hour pausing, etc...) advanced users are able to pause the print, edit the gcode, reheat the extruder/bed, and resume printing from where it left off.

Example gcode for this new feature:
```
...
; Layer 1 start
; START Object 1 print
[gcode]
...
; END Object 1 print
; START Object 2 print
[gcode]
...
; END Object 2 print
; Layer 2 start
; START Object 1 print
[gcode]
...
; END Object 1 print
...etc...
```

This would allow the gcode file to be put through a post-processor to strip out one object in the file, and resume the print job where it left off. Stripping would be done by specifying "Object 1" as an argument, and the post processor would remove all gcode between `; START Object 1 print` and `; END Object 1 print`.

Some customization could be given for the name of the label (i.e. "Object", "Item", etc...), so that it doesn't interfere with other labels in the gcode.

bubnikv, 2017-11-28:
Implemented by https://github.com/prusa3d/Slic3r/commit/672194b475e2ec6d25ed2cc3e6164144a9f7d84e and https://github.com/prusa3d/Slic3r/commit/9ca63f16bc23207c06ce51e2486fc9ce5036114f

Based on the work by @lordofhyphens
https://github.com/prusa3d/Slic3r/commit/672194b475e2ec6d25ed2cc3e6164144a9f7d84e

There is now a custom between_objects_gcode block, which you may use to implement your markers. A new variable current_object_idx may be placed into the custom G-codes, which is a zero based index of the sequentially printed objects. If the current_object_idx is placed into between_objects_gcode, the first current_object_idx application will produce an index 1, as it is a transition from the object 0 to object 1.

closed 2017-11-28 by bubnikv

nonchip, 2022-01-01:
gotta love it when stuff like this that's been implemented for 5 years never made it in the docs... there's literally a page on the prusa website called "**all** gcode template variables", and the only thing related to this it even mentions is the `complete_objects` boolean from the config file...