prusa3d/PrusaSlicer #579: version of Slic3r Prusa edition or PrusaControl running on linux 32 bits ?

openelearning, 2017-11-20:
Hello,

I'm trying to make possible to deploy open source and open hardware nano computer (cf. https://github.com/prusa3d/Slic3r/issues/564 ) and 3D printing for schools in France. Using the old computer that are in schools could be also a complementary way. And some of them are not 64 bits.

Actually it seems that there is no version of Slic3r Prusa edition (https://github.com/prusa3d/Slic3r/releases/tag/version_1.37.2) or PrusaControl of the latest stable for linux 32 bits.

### Operating system type + version
Using Debian 8 or 9 i386.

### Behavior
* _ the problem_
I don't know how a children using stl files could make easily gcode for Original Prusa i3mk2 /mk3 with an Debian 8 or 9 i386.
* _Expected Results_
The children could have a nice gcode that runs on Prusa i3 MK2/3 using an old computer :)

Is it possible to have a version of Slic3r Prusa edition or PrusaControl running on Debian or Ubuntu i386 ?

Thanks for your attention and help.

openelearning, 2017-11-20:
Note : if we can use OBS for building Slic3r Prusa Edition, maybe this can solve also this problem (cf. https://github.com/prusa3d/Slic3r/issues/564#issuecomment-345562220 ) ?

bubnikv, 2017-11-28:
For Debian, you should be able to use the Slic3r Prusa Edition from the official Debian repository:
https://packages.debian.org/sid/slic3r-prusa
https://packages.debian.org/sid/i386/slic3r-prusa/download
Does that work for you?

openelearning, 2017-11-29:
Hello,
Thanks, I have installed the Slic3r Prusa Edition from the official Debian repository with an debiian stech customized for children education (primtux).
It seems that the command line version works but not the gui : 
`~$ slic3r-prusa --gui
Can't locate Wx.pm in @INC (you may need to install the Wx module) (@INC contains: /usr/bin/lib /usr/lib/slic3r-prusa/ /etc/perl /usr/local/lib/i386-linux-gnu/perl/5.24.1 /usr/local/share/perl/5.24.1 /usr/lib/i386-linux-gnu/perl5/5.24 /usr/share/perl5 /usr/lib/i386-linux-gnu/perl/5.24 /usr/share/perl/5.24 /usr/local/lib/site_perl /usr/lib/i386-linux-gnu/perl-base) at /usr/lib/slic3r-prusa/Slic3r/GUI/2DBed.pm line 10.
BEGIN failed--compilation aborted at /usr/lib/slic3r-prusa/Slic3r/GUI/2DBed.pm line 10.
Compilation failed in require at /usr/lib/slic3r-prusa/Slic3r/GUI.pm line 9.
BEGIN failed--compilation aborted at /usr/lib/slic3r-prusa/Slic3r/GUI.pm line 9.
Compilation failed in require at (eval 158) line 1.
`
Is there anyone that have this problem on debian strech 32 bits ?

openelearning, 2017-11-29:
Note : if I run Slic3r Prusa Edition from the menu, nothing happens.

bubnikv, 2017-11-29:
You may miss some dependencies. Ideally the packages would resolve these
dependencies automatically, so this is something to be reported to the
debian maintainer of the package.

You may check the list of dependencies for your particular Debian distro
here:
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux

eyal0, 2017-11-29:
@openelearning These are the instructions that I used for compiling Slic3r on a RaspberryPi with OctoPi: https://github.com/OctoPrint/OctoPrint-Slic3r/wiki/How-to-install-Slic3r-on-RPi

@bubnikv No, no GUI.  The Raspbery Pi runs just OctoPrint which is basically headless.

bubnikv, 2017-11-29:
@eyal0 Thank you, this is very cool. We may get somebody one day to set up a build system for Raspberry PI.

By the way, how slow is R-PI with the fully parallelized Slic3r Prusa Edition?

openelearning, 2017-11-30:
@bubnikv 
thanks for your reply. It was libopengl-perl & libwx-glcanvas-perl that was missing for me. Now it works. It's possible that wx-common is also missing (for me it was already installed) cf. https://forum.lulzbot.com/viewtopic.php?t=5437
I have sent a mail to to the debian maintainer of  slic3r PE.

openelearning, 2017-11-30:
@eyal0 thanks a lot for you help, I will add it there, and add new information about arm & slic3r PE : https://github.com/prusa3d/Slic3r/issues/564

openelearning, 2017-11-30:
@bubnikv do you know if I can close now this thread or it's better that we wait that some one will maybe correct the dependencies ?

bubnikv, 2017-11-30:
I am closing this thread. Now the ball is at the debian maintainers. Thanks for your time.

closed 2017-11-30 by bubnikv

eyal0, 2017-11-30:
@bubnikv By the way, I never got back to you on the speed issue.  Slicing on RPi is pretty slow.  I never compared before/after Intel TBB so I couldn't say if the threading helped.  Maybe if I have time one day I could run a version before TBB against one right after and compare...?

I'm still unable to compile >=1.38.0 on RPi, by the way, because I think that it requires a newer libboost which I don't have.  1.37.2 is working fine, however.

bubnikv, 2017-12-01:
@eyal0 
What is the latest boost version distributed with raspberry PI?
What are the error messages regarding boost?

> Slicing on RPi is pretty slow.

I estimate it will be about 10x slower than on my 2 years old 4 core i7 laptop.

I don't have time to go through the complete installation process, but I will try to help you as much as I can. It would be great, if you could create an installable package for the raspberry PI for the community, for which I don't have time.

eyal0, 2017-12-01:
I'm not sure that it's a libboost issue, just guessing.

```
[ 11%] Building CXX object xs/CMakeFiles/libslic3r_gui.dir/src/slic3r/GUI/Preset.cpp.o
/home/pi/Slic3r-1.38.2/xs/src/slic3r/GUI/Preset.cpp: In member function ‘void Slic3r::PresetCollection::load_presets(const string&, const string&)’:
/home/pi/Slic3r-1.38.2/xs/src/slic3r/GUI/Preset.cpp:254:66: error: ‘begin’ was not declared in this scope
  for (auto &dir_entry : boost::filesystem::directory_iterator(dir))

```

Latest libboost is 1.55 on my RPi but maybe there are repositories with later versions that I don't have?

bubnikv, 2017-12-01:
I think the last entry on the following page will solve your issue. If you
could please test it and provide a pull request, I will merge it.
https://stackoverflow.com/questions/40241163/boost-directory-iterator-begin-and-end-not-declared

eyal0, 2017-12-02:
@bubnikv [This commit](https://github.com/prusa3d/Slic3r/commit/af6eae7317a4c0db13fd80c9088bc2a60d7f9198) would fix the problem.  But I don't think that you should do it.

I was able to install newer libboost on my Raspberry Pi and compile the latest sources.  It wasn't that hard.  The commit above makes the code uglier just to support an old library when it would be better to encourage users to update their boost library.

So I think that you should just leave the code as is and I'll put updating libboost into my script.

hyperair, 2017-12-04:
@openelearning 

> thanks for your reply. It was libopengl-perl & libwx-glcanvas-perl that was missing for me. Now it works. It's possible that wx-common is also missing (for me it was already installed) cf. 

The packages you mention (libopengl-perl and libwx-glcanvas-perl) are in the "Recommends" list rather than "Depends" as slic3r-prusa can be run without its GUI component, and does not need those two packages when run without GUI. If you install slic3r-prusa normally (without `--no-install-recommends`), you would have those packages installed already.

bubnikv, 2017-12-04:
@hyperair Thanks for the explanation. I am an inexperienced Debian user, but it seems to me confusing that the user will not get the GUI parts installed by default. Is there a way to flip the defaults, so the GUI components will be installed by default and the command line only solution will be optional?

openelearning, 2017-12-04:
@hyperair 
Thanks a lot for your feedback. I think I didn't install slic3r PE with this option, but with synaptic on debian strecth 9 i386. Do you want I confirm that when you install slic3r with synaptic on debian strech i386 these 2 packahes will not be installed ? I read that on the web other people have meet this problem.

openelearning, 2017-12-04:
An other way could be make a slic3r PE no GUI package, and a slic3r PE GUI package ?

hyperair, 2017-12-05:
> @hyperair Thanks for the explanation. I am an inexperienced Debian user, but it seems to me confusing that the user will not get the GUI parts installed by default. Is there a way to flip the [...]

They should get installed by default, that's what "Recommends" are. They only don't get installed by default if the user has explicitly told apt not to do so, either via the `--no-install-recommends` option, or through some apt config in /etc/apt/conf.d.

hyperair, 2017-12-05:
Some resources to read on the matter:
https://unix.stackexchange.com/questions/122289/why-install-recommends-default-is-true
https://lists.debian.org/debian-devel-announce/2007/08/msg00000.html

hyperair, 2017-12-05:
@openelearning, I believe you mentioned you are an installation on some sort of nano computer. Did you disable the installation of recommended packages to save disk space?

openelearning, 2017-12-05:
@hyperair ok thanks for your feedback, I didn't disable voluntary the recommended packages to save disk space but there is maybe 2 possibility that could explain that libopengl-perl & libwx-glcanvas-perl that was missing for me : 
* the distribution of debian that I use, primtux, is specially designed for old computer, so it's possible that the designers of primtux disable it for saving disk space ;
* and/or when I have installed slic3r it's possible that I have manually selected recommended packages (I thought they will not be installed) so maybe selecting some recommended packages disable the others recommended packages.
If I understand better the origin I will give new more news about it.