prusa3d/PrusaSlicer #580: When slicing with 0.15 mm for multimaterial it makes solids surfaces in between two STL files - this is not happening with 0.2 layer height setting

henrikkoch, 2017-11-20:
Using slicer 1.37.2 PE 

When I slice with 0.15 it makes 7 layes of solid in between to the next STL files filament when printing. You can see it on the attached images. 

If using 0.2 layer height you can see the selected pattern also in the layer between the two different filaments shown as white and yellow on my images. You will find the wrong when using 0.15 and the go to 7.25 mm height - here you will find a solid layer between. 

Is it not allowed to use 0.15 for multi material?

Also attached the two STL files and my slicer config file
[spire 3D files.zip](https://github.com/prusa3d/Slic3r/files/1489422/spire.3D.files.zip)

[image]
[image]

bubnikv, 2017-11-28:
I think the problem is, that the two STLs do not touch each other, but there is a gap between the top of the frame and the bottom of the face. Depending on the layer height, Slic3r may or may not detect this gap.

If you just lower the face by 0.1mm, I believe your issue will disappear. Also it is safe to let the two bodies intersect a bit.

bubnikv, 2018-02-12:
No response received, closing.

closed 2018-02-12 by bubnikv