prusa3d/PrusaSlicer #581: Wrong printing when sending to prusa printer and start printing from Octoprint

henrikkoch, 2017-11-20:
Using slicer 1.37.2 PE 

When having a dual STL file print and sending this to the printer over network and start printin this from Octopi - it uses the same filament in both primes. 

When exporting the file instead and copying this to SD card and printing from the printer - it works fine and starts priming using both different filaments?

Here the fwo STL files and my config file from Slicer
[spire 3D files.zip](https://github.com/prusa3d/Slic3r/files/1489428/spire.3D.files.zip)

bubnikv, 2017-11-28:
Thank you for the report. This is most likely a firmware issue and it needs to be addressed by @PavelSindler  in the Prusa3D firmware.

bubnikv, 2017-12-13:
@repetier

FYI, starting with the build 1.38.4, we changed the directory structure of our Slic3r PE slightly.
First, the default configuration is Slic3rPE, not Slic3r (actually this is the default enforced by Repetier),
but more importantly, the print, filament and printer presets were moved to a Slic3rPE/presets directory.
This change has been made to separate the presets from the things to follow, mainly the temporary data to be downloaded from the internet, like the preset updates.

Would you please create a new Repetier build with these adjustments?

Thanks, Vojtech

repetier, 2017-12-13:
Thanks for letting me know the changes. Will think about this when bundling the new version.

For the presets we used already a different directory that we set with the --datadir option . Does this still work is that now also interpreted differently? I mean until now everything was in datadir/[printer|print|filament] and now you say it moved to a preset subdirectory of SLic3rPE which sounds like not only the base dire changed (which is good to separate from slowly developing original) but also mention a presets subdir or did I understand that wrong and you mean the 3 subdirs with that?

bubnikv, 2017-12-14:
@repetier 
> Thanks for letting me know the changes. Will think about this when bundling the new version.

Please forget about the new "presets" subdirectory for the print, printer and filament profiles. There were practical reasons for the decision, but after a couple of nights of sleep I consider the change not a good idea. Mainly, it will make it difficult for people to switch between the upstream and our Slic3r using just the --datadir. I will roll back that change.

So the --datadir option stays, the only change is the default name of the datadir, which will be Slic3rPE for our Slic3r, which is the directory you call our slic3r with the --datadir already

Sorry for the confusion.

Now I have to retest, whether our latest build will work with the latest Repetier.

repetier, 2017-12-14:
That is great. By default we use Roaming/Slic3rPE for the PE version already through datadir option, so I like that choice. For an external slic3r you can select any name so that should be no problem, too. Only when I bundle it I need to make the change of executable name, which is no problem. SO this solution is compatible to existing versions.

bubnikv, 2017-12-14:
@repetier 
By the way, where do you store the source AMF and config, and in which directory do you expect the result?

By the way, both upstream Slic3r and Slic3r Prusa Edition now implement a notion of "compatible_printers". The print and filament profiles may be marked with a list of printer profiles, that are compatible with the active printer profile. If the compatible_printers field is missing in the print or filament profile, that means, that the profile is compatible with everything. Otherwise Repetier should suppress those incompatible print and filament profiles based on the active printer profile. I hope it is clear. For the Prusa3D printers, it will be important now, if you could please support that filtering, as we merged our profiles for the MK2, M2 Multi Material, and the new MK3 printer into a single config directory structure and we rely on the "compatible_printers" feature to control the clutter.

repetier, 2017-12-14:
Host has a own working directory normally in AppData/Local/RepetierHost i think it was. There we generate amf and assembled config.ini but we send full path to slic3r so that should be no problem. In log file you even see the command line for replay.

Will check the compatible printers handling and implement it. Really like that as it was cluttering my folders with so many different configs I have.

repetier, 2017-12-14:
Let me know if the fixed version is out. Might be a reason to push a host version:-)
So I can directly add the changes.

bubnikv, 2017-12-14:
@repetier
There is one more thing I just found out, which is important to make it
work.
The new custom g-code placeholder parser offers the printer_preset,
filament_preset and print_preset config keys to control the G-code macro
processing. We use it in our upcoming profiles to control the linear
advance. Currently the Slic3rPE fails to slice from Repetier (meaning it
fails to slice from a command line with our settings) due to failing G-code
post processor. I may need to extend the command line Slic3r to read these
values from config.ini, and you would need to extend Repetier to store
these keys.

bubnikv, 2017-12-14:
@repetier BTW I hijacked a wrong github issue thread, sorry to all.

Anyway, please don't remove the temporary data. When generating a G-code, I am writing a gcode.tmp first, which I then rename to gcode. If the G-code macro processing fails, the errors are written into the gcode.tmp, so if you delete it, there is no way to debug it.

[image]

repetier, 2017-12-14:
We do not delete anything in workdir, so that should be no problem. Error just comes up because the composition.gcode is not there which would be there if it would have succeeded. But the tmp file mentionted should be there if it writes in same folder.

Can I test the printer_preset, filament_preset and print_preset with 1.38.4 or do I need to wait? And also is there a documentaion? At the moment I do not really understand what you mean so I need to see it in action I guess.

closed 2018-07-19 by vojtechkral