prusa3d/PrusaSlicer #582: Supports not generated under some hanging parts

torandi, 2017-11-20:
### Version
1.37.1-prusa3d-win64

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
I have encountered this with multiple models, where supports are not generated in some places where there's nothing to support the lowest point of a part of a model hanging down, resulting in the printer starting to print in the air.

[image]

#### STL/Config (.ZIP) where problem occurs
[missing_supports.zip](https://github.com/prusa3d/Slic3r/files/1489460/missing_supports.zip)

bubnikv, 2017-11-28:
Thanks for the report. This is a known issue and it will be addressed.

p3ck, 2018-02-09:
I see this as well.  any news?  Thanks

bubnikv, 2021-02-22:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1122
See https://github.com/prusa3d/PrusaSlicer/issues/1122 , there is now support generated below the statue coat tips.

closed 2021-02-22 by bubnikv