prusa3d/PrusaSlicer #583: Feature request: Please implement Coasting

ctzsnooze, 2017-11-21:
Hi

I'd be *so grateful* If a coasting function could be implemented. 

Very helpful for preventing blobbing after travels.

Just make the extruder stop for a user specified distance before a travel, with perhaps a setting to ignore travels below a certain length.

I can confirm that this setting makes a huge difference to prints made in Simplify3D.  It's the main 'feature' missing in Slic3r Prusa edition.

Thanks for considering this and all your great work!

Chris

Itox001, 2017-11-28:
I agree, it's a simple feature that could make a great difference in quality. I actually asked for it a while ago, @bubnikv actually answered, but I think he never got around to it... Issue #246

bubnikv, 2017-11-28:
Closing this issue as it is a duplicate of https://github.com/prusa3d/Slic3r/issues/246
We may implement coasting, but the feature priority is driven primarily by our customer needs. I hope you understand that.

closed 2017-11-28 by bubnikv

DarkMonkChad, 2019-10-21:
Any chance there is an update on this? (also thank you for all of your hard work on this)

 I am printing TPU and find that I could REALLY use coasting at the end of a perimeter before a retract. I am using a .8 nozzle and if it sits still at all you get ozze. I am retracting 5mm, which helps but introduces air in to the nozzle and I am getting a consistent bubble/pop a set distance after the next extrusion starts. I believe coasting would solve this but I use to use Simplfy3D a lot and am use to using it there. If there is another answer to this, please let me know. Thank you.