prusa3d/PrusaSlicer #584: Wrong Extruder is used in Single-Material mode on Prusa MM upgrade.

normcushing, 2017-11-22:
### Version
1.37.2-Prusa3d 

### Operating system type + version
Mac OS Sierra, 10.12.6

### Behavior
Wrong Extruder is used to print when 
[slic3r_stl_and_config_ini.zip](https://github.com/prusa3d/Slic3r/files/1493621/slic3r_stl_and_config_ini.zip)
operating the Prusa MK2S with Multi-Material upgrade in Single-Material mode (i.e., using the provided Printer Settings config, "Original Prusa i3 MK2 MM Single Mode").

The Slic3r config bundle loaded is:
mk2mm_Slic3r_prusa3d_1_9_1.ini

The firmware in the printer is:
1_75mm_MK2-MultiMaterial-RAMBo13a-E3Dv6full.hex

Printer Settings config is:
Original Prusa i3 MK2 MM Single Mode

PROBLEM - BUG:
When the print starts the Prusa printer display prompts for the Extruder to use when making the print. If the last extruder used was E1 and E2 is selected as the new extruder the printer displays E2 as expected. It then proceeds to lay down a strip of material to flush the old filament from E1 and to prime the system with filament from E2. When that task is finished the print head moves to print the object. The prusa printer display changes to show extruder E1 (the wrong extruder). And, it proceeds to print while trying to use E1. Unfortunately, the printer is now trying to use the wrong filament via the wrong extruder, and nothing is printed because this issue causes a jam because the E2 filament is loaded into the hotend following the flush and priming task. Thus E2 filament is blocking E1 filament from extruding. Please realize it was E2 filament that is desired and not E1 filament. The jam must be corrected before printing can continue.. I have not found a work-around.

NOTE: The Prusa Control app gcode for a Single Material print performs as expected. It flushes/primes and prints with the selected extruder.

NOTE: Both the Prusa Control app and the Slic3r app gcode do not automatically unload the old filament/extruder once a different filament/extruder are selected. When using the Prusa Control app and Slic3r app gcode I must manually unload the old filament/extruder before beginning the print using the new selected filament/extruder If this step is not done a jam will occur at the beginning of the flush/prime task.

ADDITIONAL ENHANCEMENT - NEW FEATURE REQUEST:
When the firmware prompts for the extruder to use, if the selected extruder is different than the active or last used extruder, it would be very nice if the printer automatically performed a partial unloaded of the old extruder before using the new extruder. By partial unload I mean it backed the filament out of the way of the new filament.

#### STL/Config (.ZIP) where problem occurs
Files should have been provided.

bubnikv, 2017-11-28:
Would you please send me the G-code?

Inside the single material G-code there should be only one line starting with T to be found, that is T?
There should be no other line starting with T contained in your G-code. If that is the case, then you will likely experience the incorrect extruder switching you reported.

I tried to export a G-code with your settings and your STL, but I got only a single "T?" line, which should be correct.

> NOTE: The Prusa Control app gcode for a Single Material print performs as expected. It flushes/primes and prints with the selected extruder.

This is interesting, but it may happen. Prusa Control calls Slic3r behind the scenes, but there may be a slightly different version of Slic3r bundled with Prusa Control and the configs could be slightly different as well. Would you please attach both the G-code generated by Slic3r standalone and the G-code generated by the Prusa Control?

> NOTE: Both the Prusa Control app and the Slic3r app gcode do not automatically unload the old filament/extruder once a different filament/extruder are selected. When using the Prusa Control app and [...]

The unloading of the filament at the end of the print is performed by the end G-code. This G-code section will shape the end of the filament and it will park the filament end in the cooling tube, so the next print should start flawlessly. If that does not work for you, then either the filament shaping sequence does not work with your filament material, or your software install is messed up.

> ADDITIONAL ENHANCEMENT - NEW FEATURE REQUEST:
When the firmware prompts for the extruder to use, if the selected extruder is different than the active or last used extruder, it would be very nice if the printer automatically performed a partial unloaded of the old extruder before using the new extruder. By partial unload I mean it backed the filament out of the way of the new filament still leaving it loaded into the multiplexer as is done on a "Load". If this is done, it may require the "Select Extruder" to move the filament forward as if it was not partially unloaded so that filament for the flush/prime is immediately available.

This should be working already in case the preceding print finished correctly. Also the print cancel command of a MM printer firmware should be doing the same filament shaping and parking sequence, so that should work as well.

Would you please share the details of your filament?

bubnikv, 2017-12-14:
I think in some cases you used a wrong profile for your Multi-Material printer.

Also I agree that there was an issue with the Slic3r "Original Prusa i3 MK2 MultiMaterial" profile as noted: If an object was printed with a single extruder only, then a wipe tower was not generated and the printing filament was not pulled and parked into the cooling tubes at the end of the print. That has been fixed in Slic3r 1.38.4. Please note that you will need the updated Slic3r profiles for this change to work. You may just run the wizard and install the newly bundled Slic3r profiles.

normcushing, 2017-12-17:
Hi,

I apologize for the longish delay in replying. Daily affairs needed attention.

Per your request attached is the gcode file, disk_mm_v4.gcode, that demonstrates the problem. I ran some additional tests using the gcode file and achieved the same bad results as when the bug report was issued. As you mentioned in the first email, this gcode file does contain the “T?” line. Notice it also contains comments for settings for four extruders. Is that expected?

Using the disk_mm_v4.gcode file here is what I did to duplicate the problem.

1st print
	Print from SD
	I selected the file disk_mm_v4.gcode.
	I was prompted to choose extruder.
	To get a baseline that the file worked I selected E1 (no change in extruder from how the gcode file was built).
		LED stated E1 during bed leveling.
		Pre-print priming acted like an MM printer but in SM mode. Meaning a 250mm run of priming material was extruded.
		It printed the model.
		At end of printing it cleared E1 filament from the hotend.
		No filament remaind in the hotend.
	This was good.

2nd print
	Print from SD
	I selected the file disk_mm_v4.gcode.
	I was prompted to choose Extruder. I selected E2. 
		LED stated E2 during bed leveling.
		Pre-print priming acted like an MM printer but in SM mode.
			Color transitioned from the E1 to the E2 color as it laid down a 250mm strip of priming material.
		When printing LED switched back to E1.
		Nothing was printed.
		At end of printing it attempted to clear E1 filament from the hotend.
		At end of the print the filament for E2 was in the hotend.
	This was bad. Nothing was printed and filament was jammed.

A fresh attempt to duplicate the problem. But I could not duplicate it:

I have not yet downloaded and installed the updated firmware, Slic3r 1.38.4., you mentioned in the reply.

I created another test file using fresh settings to start over and try to recreate the problem. I was not able to duplicate the problem. It worked fine. The stl and gcode file for this test are attached: fourSides_v1.stl and fourSides_e1_v1.gcode

When using the slic3r prusa edition software version 1.37.2 Prusa3d…
	From the Settings, I selected Extruder 1.
	When the object was printed it prompted me for the extruder to use.
		I selected an extruder other than E1 and it worked fine.
		I ran it many times selecting other extruders and it worked each time. 
	It always unloaded the filament at the end of the print thus no jams occurred when the Extruder was changed.

Something to consider:
Could something have changed in my system that corrected the problem? In other words, I am wondering if reloading the config bundle files (mk2mm_Slic3r_prusa3d_1_9_1.ini) might have only partially worked initially and with a reloading the values were properly set? I bring this up because at some point in time I noticed the when switching between Printer settings of Mutli-Mode and Single-Mode the wipe tower did not go away as anticipated. At the time I thought it was weird but did not have enough experience to explicitly know. Today when I change modes the tower comes and goes as expected.

-Norm

> I think in some cases you used a wrong profile for your Multi-Material printer.
>
> Also I agree that there was an issue with the Slic3r "Original Prusa i3 MK2 MultiMaterial" profile as noted: If an [...]

-norm

norman cushing
Norman Cushing Productions
520-425-1516
business@normancushing.com

normcushing, 2017-12-17:
Hi,
Per your request to know the filament, I was using Hatchbox PLA in a variety of colors. I have printed many multi-mode prints using the material with great success that includes 2 to 4 color changes per layer so I believe the filament is not the issue.
-Norm

> Would you please send me the G-code?
>
> Inside the single material G-code there should be only one line starting with T to be found, that is T?
> There should be no other line starting with T contained [...]

-norm

norman cushing
Norman Cushing Productions
520-425-1516
business@normancushing.com

bubnikv, 2017-12-18:
> Notice it also contains comments for settings for four extruders. Is that expected?

This is a heritage from the original Slic3r. The Slic3r 1.38.4 works correctly with the single extruder multi material setup and it will only emit G-code for the active filament.

> Could something have changed in my system that corrected the problem? In other words, I am wondering if reloading the config bundle files (mk2mm_Slic3r_prusa3d_1_9_1.ini) might have only partially worked initially and with a reloading the values were properly set?

I cannot exclude that, but it is weird.

> I bring this up because at some point in time I noticed the when switching between Printer settings of Mutli-Mode and Single-Mode the wipe tower did not go away as anticipated. At the time I thought it was weird but did not have enough experience to explicitly know.

Every time you think there is something wrong going on, it would be great, if you could try to reproduce the situation and describe as precisely as possible for me how to reproduce it. It may be, that something did not update correctly when you switched from the Multi-Mode to Single-Mode. I have reworked a lot of that code in 1.38.4 (I rewrote the configuration layer from Perl to C++), which means these issues may be gone, or there may be another new set of issues (hopefully not).

Thank you.

bubnikv, 2018-03-14:
We have fixed one issue in 1.39.1: When printing a single material print with the multi-material Slic3r profile, the Slic3r did not pull the filament into the cooling tubes. We expected that one will print the single material prints on the multi material printer with the "MM Single Mode" profile, so we mistakenly did not include the parking of the filament into the cooling tubes for the "MM Multi" profile if printed with a single material only. This has been fixed in 1.39.1 with the conditional G-code, where the same parking sequence is now used for the MM-single as for the MM-multi profiles.

closed 2018-03-14 by bubnikv