prusa3d/PrusaSlicer #586: Problem with layer editing

Khenals, 2017-11-23:
### Version
-1.37.2

### Operating system type + version
-Windows 10 64 bits

### Behavior
-When I tried to used the layer editing I got this error, help pls?
"Cannot compile an OpenGL Shader, therefore the Variable Layer Editing will be disabled. Windows is using a software OpenGL renderer, You are either connected over remote desktop or a hardware acceleration is not avalaible. 
GL version: 1.1.0 
vendor: Microsoft Corporation
renderer: GDI Generic
GLSL version: N/A"
Any solution?

bubnikv, 2017-11-24:
For some reason, Windows provided Slic3r with the software OpenGL renderer. Does your graphics card support OpenGL 2.0?

bubnikv, 2018-02-12:
No response, closing.

closed 2018-02-12 by bubnikv