prusa3d/PrusaSlicer #587: Avoid Crossing Perimeters - strange counter-effective behavior with non-trivial design

scalablecory, 2017-11-24:
### Version
1.37.2

"Avoid Crossing Perimeters" is causing some grief for one of my prints, where it is causing stringing and increasing print times with unnecessary travels.

Here it is with a large travel in between fine details when a simple short move would be fine.
[image]

Here it is printing a single perimeter wall, moving away to print something tiny, and then moving to do the next perimeter wall:
[image]

Here is the layer that is creating this bad behavior:
[image]

And the STL/config:
[object.zip](https://github.com/prusa3d/Slic3r/files/1500626/object.zip)

scalablecory, 2017-11-24:
Video links in case gif is too choppy:
https://gfycat.com/gifs/detail/InsignificantBasicFulmar
https://gfycat.com/gifs/detail/MajorFavoriteGraysquirrel

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

[image]

closed 2020-12-11 by bubnikv

nordurljosahvida, 2020-12-12:
Fantastic thank you