prusa3d/PrusaSlicer #588: MMU Wipe tower:  Fan not activating when bridging over sparse infill

domesticatedviking, 2017-11-24:
### Version 
1.37.2-prusa3d-linux64

### Operating system type + version
Ubuntu 16.04LTS

### Behavior
When printing wipe tower, part cooling fan does not activate when bridging over sparse infill.  This results in an unusable wipe tower - impossible to print in ABS.    I do not believe this is a hardware issue since my part fan operates as expected when activated manually via terminal with M106.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[Bridging not working on wipe tower.zip](https://github.com/prusa3d/Slic3r/files/1502365/Bridging.not.working.on.wipe.tower.zip)

domesticatedviking, 2017-11-25:
On further investigation, it seems that the issue may be that auto cooling settings work differently than I assumed.   Suggest the UI be clarified and tooltips be improved to explain that "Bridges fan speed" is applied as a percentage of  "Max" Fan speed.    Personally I think the bridges fan speed should be able to be set independently.

domesticatedviking, 2017-11-25:
Reopening this as problem persists.  Fan is not activating on bridge 5.4mm into the model attached in my initial post.  Wipe tower is unusable as a consequence - print fails unless I manually activate fan during bridging.

domesticatedviking, 2017-11-29:
I have tested my hardware thoroughly.  Other models that i slice will activate my part fan automatically where there are bridges.  The wipe tower bridge doesn't seem to get picked up though.
[image]

bubnikv, 2017-11-29:
@domesticatedviking 
> I have tested my hardware thoroughly. Other models that i slice will activate my part fan automatically where there are bridges. The wipe tower bridge doesn't seem to get picked up though.

You are correct. The bridging has not been implemented correctly for the wipe tower. Neither the bridging flow nor the cooling is applied to the bridges over the wipe tower. We have now a new hire working on the wipe tower, so hopefully it will get his attention.

domesticatedviking, 2017-11-29:
@bubnikv  Hey - thanks!  Glad to know I'm not going crazy :)

domesticatedviking, 2017-12-14:
Any way to provoke that new hire to get on this?   This is a major issue that completely breaks the MMU's compatibility with ABS.

domesticatedviking, 2019-05-26:
Closing this issue this is no longer a problem.

closed 2019-05-26 by domesticatedviking