prusa3d/PrusaSlicer #589: Extruder Clearance should be in printer settings, instead of print settings

echo465, 2017-11-25:
### Version
Slic3r-1.37.2-prusa3d-win64-full-201710211408

### Operating system type + version
Windows 7 64

### Behavior
The sequential printing section under Output Options in the Print settings tab prompts for Extruder clearance and Height.  These measurements are dependent on the printer hardware, so if I want to send the same job to 2 different printers, I need to change these setting here rather than have them preset in the printer settings tab.

fornellas, 2020-11-17:
@rtyr I can give it a try to do a PR for this. Would you be OK with moving this setting to `Printer Settings > General > Size and coordinates`?

Also, if you have any pointers for me to be aware (I'm new to the codebase), thanks in advance for sharing.

Area5142, 2020-11-19:
As this is a physical thing, maybe the setting should be at the physical printer as one printer can be modified to have more clearance than another of the same model.

closed 2025-02-18 by rtyr