prusa3d/PrusaSlicer #591: [Feature Request] Ability to update profiles from Slic3r

devcompl, 2017-11-28:
### Version
Slic3r Prusa Edition
Version: 1.37.2-prusa3d-win64
Build: Slic3r-1.37.2-prusa3d-win64-full-201710211408
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 10.0.16299.19] 

### Behavior
Given how fast the profiles change at [Slic3r-settings](https://github.com/prusa3d/Slic3r-settings) repo it would be nice if we had an updater build-in in Slic3r. It would detect any changes to the default profiles and prompt user to update. Maybe even show the changes?

Profile version could be the date in the "generated by Slic3r" header in profile files.

What do you think?

bubnikv, 2017-11-29:
This is a similar issue tohttps://github.com/prusa3d/Slic3r/issues/354
We would like to have a read only "OEM" profile, which will be downloaded automatically. Some of the format has been discussed at the end of https://github.com/prusa3d/Slic3r/issues/395

bubnikv, 2018-02-12:
@vojtechkral is currently working on updating the user profiles from the internet.

bubnikv, 2018-08-02:
It is working for some time now. Closing.

closed 2018-08-02 by bubnikv