prusa3d/PrusaSlicer #592: Repository for images, for Wiki and releases

bubnikv, 2017-11-29:
As github does not offer a better repository for images, this issue will be used as one. This issue will be closed right away to be hidden.

[image]

bubnikv, 2017-12-20:
Reopened, otherwise github seems to delete the images after a year.

bubnikv, 2017-12-20:
Uploaded images for the wiki and release logs.
[image]
[image]
[image]

bubnikv, 2018-09-26:
[image]

bubnikv, 2018-09-26:
[image]

bubnikv, 2018-09-26:
[image]

bubnikv, 2019-08-08:
[image]

[image]

[image]

[image]

[image]

bubnikv, 2019-08-08:
Animated gifs. Are these supported by github?

[image]
[image]
[image]

bubnikv, 2019-08-09:
[image]

bubnikv, 2020-10-13:
SLA tree support improvements in problematic parts of meshes, like holes and cavities
[image] | 
[image]

Import the profile used for printing and/or the full reconstructed mesh from SL1 output archives.
[image]
[image]
[image]

Improved SLA support point generation with better overhanging edge and hollowed interior handling.
[image]
[image]
[image]
[image]

Improved SLA rotation optimizer that supports printing directly on bed.
[image]
[image]

Hollowing improved for meshes with separate watertight parts. (may I say multi-part here? also partly in-progress)
[image]
[image]

lukasmatena, 2020-10-16:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

bubnikv, 2020-10-17:
[image]

mikolaszuza, 2021-08-17:
Fuzzy skin
[image]

mikolaszuza, 2021-08-18:
Negative volume
[image]

mikolaszuza, 2021-08-18:
Did you know tooltip
[image]

File transfer
[image]

mikolaszuza, 2021-08-18:
Simplify mesh
[image]

lukasmatena, 2021-08-19:
[image]

mikolaszuza, 2021-08-30:
Multi material painting video
https://user-images.githubusercontent.com/23498906/131332364-40eed6ee-4e00-4731-9752-87f0752c980b.mp4

mikolaszuza, 2021-08-30:
Snug vs Grid supports
[image]

lukasmatena, 2026-09-04:
Probably safer to keep this one open, as noted above: https://github.com/prusa3d/PrusaSlicer/issues/592#issuecomment-353004968