prusa3d/PrusaSlicer #594: Can't run .AppImage on ubuntu 17.10

JesusFreke, 2017-11-29:
### Version
1.38.2-prusa3d-linux64-full-201711292009.AppImage

### Operating system type + version
Ubuntu 17.10

### Behavior
When I try to launch the .AppImage from https://github.com/prusa3d/Slic3r/releases, it immediately fails with an error.

    $ ~/bin/Slic3r-1.38.2-prusa3d-linux64-full-201711292009.AppImage 
    Can't locate object method "GetUserDataDir" via package "Wx::StandardPaths::Get" (perhaps you forgot to load "Wx::StandardPaths::Get"?) at /tmp/.mount_Slic3rlGtkX3/usr/bin/bin/slic3r.pl line 107.

I've tried installing the `wx-common` and `libwx-perl` packages, but still encounter the error.

JesusFreke, 2017-11-29:
I also get the same error when I download the tarball and try to run it that way.

aequitas, 2017-11-29:
Getting the same on macOS. 
```
$ /Applications/Slic3r.app/Contents/MacOS/Slic3r
Can't locate object method "GetUserDataDir" via package "Wx::StandardPaths::Get" (perhaps you forgot to load "Wx::StandardPaths::Get"?) at /Applications/Slic3r.app/Contents/MacOS/slic3r.pl line 107.
```

Snowflake6, 2017-11-30:
Oh, good, it's not just me doing something stupid.  #4 with same behaviour and same error message.

rmoravcik, 2017-11-30:
I have the same problem on my Ubuntu 17.10

`morgan@moravcikpc:~/opt$ ./Slic3r-1.38.2-prusa3d-linux64-full-201711292009.AppImage 
Can't locate object method "GetUserDataDir" via package "Wx::StandardPaths::Get" (perhaps you forgot to load "Wx::StandardPaths::Get"?) at /tmp/.mount_Slic3rvhpBss/usr/bin/bin/slic3r.pl line 107.
morgan@moravcikpc:~/opt$`

bubnikv, 2017-11-30:
Shame on me, I did not test on other systems than Windows this time, I was
in hurry.
I will fix it today with a patch release.

bubnikv, 2017-11-30:
I have fixed that issue with https://github.com/prusa3d/Slic3r/commit/6aff27f3abf6ea895cc458845a31bc436ffa38a8
A new alpha will be released in a day or two.

bubnikv, 2017-11-30:
reopened until a release is ready, so this issue is documented.

bubnikv, 2017-12-05:
Fixed with Slic3r 1.38.3 beta release.

closed 2017-12-05 by bubnikv

ir-fuel, 2017-12-09:
Problem still exists