prusa3d/PrusaSlicer #597: Slic3r crashes when sending STLs with special characters to the printer

lilithpro, 2017-12-01:
### Version
NOT present in 1.36.2-prusa3d-win64
appeared in 1.37.1-prusa3d-win64
also verified in 1.37.2-prusa3d-win64 & 1.38.2-prusa3d-win64

### Operating system type + version
Windows 10 Pro 1703

### Behavior
Loading STL with special character (e.g. fuß.stl (attached), slicing and saving gcode works as expected.
Clicking send to printer crashes Slic3r at the stage "Sending G-code file to the OctoPrint server..."
Error message in terminal:
`Malformed UTF-8 character (fatal) at C:/Program Files/Slic3r/lib/File/Basename.pm line 338.`
(Never mind the directory name)
Removing the special character from the stl filename works as expected.

#### STL/Config (.ZIP) where problem occurs
[demo.zip](https://github.com/prusa3d/Slic3r/files/1521932/demo.zip)

bubnikv, 2017-12-05:
What default system locale do you have selected at your Windows 10 box? If it is different from German, then Slic3r will mangle the German characters. This is a limitation of Perl interpreter running on MS Windows. This limitation will go away once we get rid of Perl.

https://www.isunshare.com/windows-10/change-system-locale-in-windows-10.html

lilithpro, 2017-12-05:
My system locale is set to German (Germany)

bubnikv, 2017-12-21:
Fixed with https://github.com/prusa3d/Slic3r/commit/f5160b7a72dcb9305e6220a88092e3bd1032387e

In addition to the upstream Slic3r, I finally managed to get the file name transferred in UTF-8 format, so your file name displayed by OctoPrint will be correct as well.

closed 2017-12-21 by bubnikv