prusa3d/PrusaSlicer #598: Multi Material Soluble Interface Slicing - Missing Layers

ryan2154, 2017-12-01:
### Version
Slic3r Version 1.37.2

### Operating system type + version
Windows 10

### Behavior
Recently when slicing objects in Slic3r I have found that the exported G-code skips layers. I tried multiple different configurations but it seems to fail at the same location consistently. For example in the g-code attached layer 332 is at a z height of 76.700mm. Layer 333 is at 117.950mm which is a difference of 41.25mm. 

This error seems to be when Slic3r exports the g-code because the preview window in Slic3r has the correct layers and z-heights. I only discovered that this could be the problem when I analyzed the gcode in various other programs such as Simplify3d.

This error exists in both Slic3r (Single and Multimaterial) and Prusa Control. I have tried changing my layer settings such as "Avoid crossing perimeters" and "Detect thin walls". These changes don't seem to change anything. This problem doesn't exist if it's a small model and the g-code export is quick. With these larger models with complicated supports it takes 5-8 minutes to export. I have included pictures, STL, Gcode, and config files in the zip folder. 

Thanks for your help!

#### STL/Config (.ZIP) where problem occurs
[Slicing Issue.zip](https://github.com/prusa3d/Slic3r/files/1523048/Slicing.Issue.zip)

bubnikv, 2017-12-01:
Hello Ryan,

> I tried multiple different configurations but it seems to fail at the same location consistently. For example in the g-code attached layer 332 is at a z height of 76.700mm. Layer 333 is at 117.950mm which is a difference of 41.25mm.

This seems to be a bug in Simplify3D 4.0. If you switch the view by line, not view by line, then you will be able to trace the "missing" layers in Simplify3D. Also you may check the "missing" layers in the repetier host.

Slic3r generates support layers, which are not necessarily synchronized with object layers. This allows Slic3r to print the support towers faster than the object, but this also makes the G-code analysis difficult to Simplify3D and other applications.

> This problem doesn't exist if it's a small model and the g-code export is quick. With these larger models with complicated supports it takes 5-8 minutes to export.

I agree that it took a considerable time to export the G-code. This is something to improve upon.

Do you have an issue printing this model? Do you still think there is an issue in Slic3r?

ryan2154, 2017-12-01:
The print fails every time. I have even watched it skip the layers.

bubnikv, 2017-12-01:
ok, I will let Jindra (the print expert guy at Prusa Research) print your model.

ryan2154, 2017-12-01:
Thank you. I sent a separate email to support. I haven't tried this particular model. The actual model that is failing is proprietary (customer file) and I am unable to share it.

bubnikv, 2017-12-05:
Sorry, but the data you sent to the support were identical to those you attached to this issue. Would you please re-send your private geometry file to our support? Thanks.

ryan2154, 2017-12-12:
Issue has been resolved. G-code seemed to be corrupted causing the issue. Have sliced a few other times and the issue seems to be resolved. Thanks for your help.

bubnikv, 2017-12-12:
Thank you.
Closing the issue.

closed 2017-12-12 by bubnikv