prusa3d/PrusaSlicer #599: Support Material generation dosn't respect min Layer height

MadMe86, 2017-12-02:
### Version
1.37.2-prusa3d-linux64
AppImage

### Operating system type + version
Arch Linux 

### Behavior
Support Material generation dosn't respect the miniumum Layer height.
At the Moment i am using an Fabrikator mini II
This small one, dosn't support smaller stepps than 0,05mm (it just dont move)

To reproduce this just generate any gcode with support and take a look in the gcode...

#### STL/Config (.ZIP) where problem occurs

Upload dosn't work atm... :-/

so here a small GCODE part

```
G1 X57.864 Y19.722 E14.73409
G1 X58.447 Y19.385 E14.75674
G1 X58.776 Y18.816 E14.77882
G1 X57.700 Y19.437 E14.82059
G1 E9.82059 F2700.00000
G92 E0
G1 Z1.212 F7200.000
G1 X58.331 Y25.244 F7200.000
G1 Z1.112 F7200.000
G1 E5.00000 F2700.00000
G1 F3600
G1 X56.143 Y23.981 E5.08488
G1 X55.035 Y24.621 E5.12787
G1 X55.035 Y27.508 E5.22488
G1 X52.043 Y29.236 E5.34102

```

Rudgas, 2018-02-04:
+1 on this, same System/Version.

It should be possible to enable layer-height synchronisation for detachable support. Consequently Contact Z Distance has to be a multiple of Layer-height. Or to go very fancy, enable variable layer height only for the last layer contacting the model, I don't see much use misaligning all layers.

This is an issue as it does not respect the minimum Layer height settings, also unchecking the "enable variable layer height" is not taken into consideration. Even setting layer height limits does not effect it.
Mechanically I fear this might create issues when the support layer height is at such odd numbers that it will interfere with the normal layers by causing misalignments.

MadMe86, 2021-10-09:
It is sadly still present 2.3.3 :-/

[Pi4caseBottom-nosupports-20211009-114714-Extr.Multipl:1.25-FLTemp:205.gcode.log](https://github.com/prusa3d/PrusaSlicer/files/7315352/Pi4caseBottom-nosupports-20211009-114714-Extr.Multipl.1.25-FLTemp.205.gcode.log)
For Example at Height 2.60 there is a support layer and the next layer on top of this is 2.97 which is not a multiple of 0.05 mm
or 0.73 and 1.11
[image]

rtyr, 2021-10-09:
I don't see any issue in your g-code.

> which is not a multiple of 0.05 mm

Why it should be? Supports are using variable layer height.

> 0.73 and 1.11

No problem, it equals to 0.38mm layer height. Your range is 0.05-0.4.

Can we close the issue?

rtyr, 2023-11-23:
Working as intended, the issue mentioned here is likely on the printer side. Closing.

FabD86, 2023-11-23:
Too bad to read that it will not be fixed.
But please mark it as won't fix and not as completed.
This issue still persists, of course only for some strange hardware.

rtyr, 2023-11-23:
If your hw needs fixed layer height for the supports, you can use organic supports. Organic supports are printed with the same layer height as main object.