prusa3d/PrusaSlicer #600: Non-ASCII characters or unknown macros in g-code cause it to be truncated

DrLex0, 2017-12-02:
### Version
1.38.3-prusa3d
This is a regression, was not present in 1.37.x

### Operating system type + version
Mac OS 10.11.6

### Behavior
* Non-ASCII characters in G-code, as well as unknown macros, cause it to be truncated right before the character or macro. To make things worse, merely typing something like “don't” in the G-code edit field, will automatically convert the single quote “'” to a non-ASCII curly quote “, and there are many other such automated conversions (these should NOT be occurring, but that's another bug I guess).
* Steps to reproduce:
  1. Copy the following two lines into the Start G-code field:
  ; This isn’t pure ASCII
  ; This is the second line of G-code
  2. Or, copy the following two lines:
  ; This is not a macro: [notamacro]
  ; This is the second line of G-code
  2. Load any model with any settings and export the G-code file.
  3. Look inside it
* _Expected Results_
  * You see both lines in the output file
* _Actual Results_
  * The file only contains: `; This isn`, or in the second case: “; This is not a macro: ”

bubnikv, 2017-12-02:
This seems to be an issue of OSX (and maybe only some versions of OSX and maybe only under some system configuration options). 

This is my note on https://github.com/alexrj/Slic3r/issues/3408 on  Jul 10, 2016:

> So this seems to be a general OSX / wxWidgets problem. Since OSX 10.9 the
smart dashes and smart quotes are on by default system wide. You may
disable them in system settings. There is currently no way to switch them
off through the wxWidgets api, but wxWidgets 3.1 disables that smart stuff
by default. Apart from the support of multisample anti aliasing, this seems
to be another reason to upgrade the wxWidgets to 3.1, but wxWidgets 3.1
break the perl wx module dependencies. I asked for help with wx module
support of wxWidgets 3.1 and got no response. I suppose the wxPerl has been
abandoned by the authors. Maybe a reasonable workaround would be to patch
wxWidgets 3.0.3 to force the smart dashes off, at least on our build server.

Would you please post some of the config.ini files with the G-codes that are processed incorrectly? I will try to convert the "smart dashes and quotes" back to the normal ones.

bubnikv, 2017-12-02:
I think I will do an ugly thing to solve this issue for good at least for our OSX builds: I will simply port the code to disable the "smart quotes" from the wxWidgets 3.1 to wxWigets 3.0.3 that we are using.

bubnikv, 2017-12-02:
There seem to be four issues going on:

Issue 1)
-----------------

wxWidgets 3.0.3 text input control on OSX applies "dash substitution" and "quote substitution" by default. This is what mangles the quotes, an apostrophe in "don't", converts ... into a single ... unicode character, and converts -- into a single long dash.

I have applied the following patch to the wxWidgets-3.0.3 on our build server:
https://github.com/wxWidgets/wxWidgets/commit/c07523734f2d93a8c9042d80fdd458954b0e1d41

Actually the two following lines need to be added to src/osx/cocoa/textctrl.mm
    [tv setAutomaticDashSubstitutionEnabled:false];
    [tv setAutomaticQuoteSubstitutionEnabled:false];

I have attached a new Slic3r PE OSX build of the same build number, but with the updated wxWidgets 3.0.3 with the substitutions disabled.
[Slic3r-1.38.3-prusa3d-full-201712022004.dmg.zip](https://github.com/prusa3d/Slic3r/files/1524129/Slic3r-1.38.3-prusa3d-full-201712022004.dmg.zip)
I have verified, that the conversion is off, and it was actually easier than I feared.

Issue 2)
-----------------

The new placeholder parser fails on unicode characters. I am using a boost::spirit parser, and I hope I can configure it to consume unicode characters correctly. It is a question though, whether such a G-code should be emitted by Slic3r, as it may not be processed by a 3D printer firmware correctly.

Issue 3)
-----------------

The new placeholder parser fails on non-existent variable names. I made the change intentionally. The new placeholder parser could process complex expressions enclosed in {}, the old [] syntax is still supported. I may modify the placeholder parser to ignore non-existent variable names enclosed in [] (the old syntax), but I would not want to allow it for the new syntax {}. I think the placeholder parser should behave consistently in both cases, it should fail. This leads to

Issue 4)
-----------------

When the new placeholder parser fails, the error report is written to a console only, which is currently not being open for Slic3r on OSX. I shall certainly improve upon it and show a useful error message in a GUI, if the placeholder parser fails.

I hope I have solved the Issue 1). I think issue 3 is not an issue really as long as issue 4 is fixed. Issue 2 should likely be fixed as well, but using Unicode characters in G-code will most likely be safe in G-code comments only.

Would you please test the attached OSX build? Please delete the "unicode" G-code section first, then re-type your G-code sections. Please verify that there is no Unicode character stored into the Slic3r ini files. I hope this Slic3r build will work for you.

bubnikv, 2017-12-02:
@lordofhyphens see the previous post, Issue 1). I am pretty sure you have the problem as well.

DrLex0, 2017-12-02:
I've tried the attached build, it indeed does not auto-replace quotes or dashes. Thanks!

As for issue 2, I wouldn't mind if any non-ASCII characters would simply be dropped from the generated G-code file, or replaced by question marks. The unicode is only in comments (and should never be outside of comments), and those comments are only to explain why certain things are in the template. Normally no human will read the generated G-code files, unless something went wrong.
I avoid non-ASCII characters altogether in any kind of code that isn't meant to explicitly output unicode or the like, but in this case I was of course stuck with issue 1… In the meantime I had a simple workaround, which was to write “we'll” or “isn't” in full: “we will” or “is not”, etc. :)

It would be nice indeed to have immediate feedback when something went wrong. In this case I only noticed when the printer started doing weird things.

bubnikv, 2017-12-03:
Thanks for the test, I have updated the build at the release page for the 1.38.3 beta. As it only differs in the patched wxWidgets class, I think it is fine to replace it at the release page.

As to the other issues, I will try to fix them in this round for the final release.

bubnikv, 2017-12-04:
I have fixed the issue 2) with https://github.com/prusa3d/Slic3r/commit/fb1bebd982c6a92ccb5217f2d0530d0bcf42ce76 and 
https://github.com/prusa3d/Slic3r/commit/1244fd09eba10f301278778dc9d10e0b752b8c76. The UTF-8 characters are allowed inside the free-form G-code including the comments, and inside string constants (enclosed in quotes).

I will yet improve the error reporting.

bubnikv, 2017-12-05:
The error reporting has been significantly improved with https://github.com/prusa3d/Slic3r/commit/8746f84fa2b8585270d36b4070693a188e722e33

bubnikv, 2017-12-05:
All the issues were fixed, the fixes will be present in the next beta. Closing.

closed 2017-12-05 by bubnikv