prusa3d/PrusaSlicer #601: Feature Request: option to disable collision detection for sequential objects

JesusFreke, 2017-12-02:
In some cases, the built in collision detection that gets used when printing sequential objects rejects prints that would actually print fine.

One simple example is a set of very short (~2-3mm) concentric rings - different sized gaskets printed with flexible filament.

The rings are modeled as a simple disk, and are all added at the same location on the build plate. I set it up to print them sequentially in vase mode. But slic3r refuses to slice with the warning "Some objects are too close, your extruder will collide with them", even when I set the extruder clearance value to very small value (e.g. .01mm).

I'm planning on making a pull request to implement an option to disable this collision detection for these types of use cases where the existing collision detection doesn't work well. 

Would you prefer this to be implemented as a new boolean option in print settings -> output options -> sequential printing? If false, it disables the collision detection functionality, and disables the other controls in the sequential printing section.

Or, a slightly simpler design might be to allow the extruder clearance to be set to 0, which would disable the collision detection.

In either case, it probably also makes sense to either show a dialog or some warning string in the settings screen saying something to the effect that collision detection has been disabled and the user must ensure the extruder won't collide with earlier objects.

JesusFreke, 2018-09-07:
Hmm. I think this feature could be used as a workaround for the problem that #1095 describes, but I agree with the author of that issue that I don't understand why that particular case is causing a collision warning. While I do understand why the particular case I describe causes a warning.

ChrisVahi, 2020-04-27:
upvote. This feature preventing slicing is annoying.

jimschaefer, 2021-01-14:
I agree.. I am trying some things I know will work and would like to be able to slice and print sequentially.  Perhaps entering zero for both the extruder clearance and height ?

AndKe, 2021-05-18:
Both issue 4270 and this is closed, and the collision detection is as drunk as ever.

fe60, 2021-06-01:
Another use case not working due to collision detection:
Sequential printing of objects with custom brim (added as separate object)
[image]
As long as there is no option to model and declare a custom brim (requested in #4744), the only option I see to get this working, is to disable the collision detection.

skarfie123, 2021-08-21:
Trying to do multicolour prints as in https://www.makeanything.design/multicolor
And the ability to turn off collision detection would be a great help!

cimbalek, 2022-12-04:
When they do not want to let you disable this workaround is: 
Radius 0
Height 1000

@jimschaefer ^

bionicbone, 2023-01-13:
> When they do not want to let you disable this, workaround is: Radius 0 Height 1000
> 
> @jimschaefer @ChrisVahi ^

But that still does not work...

[image]

[image]

[image]

I just want to print the letters in Red and the outer line in White, Letters first, Line Second... Help

rizwansarwar, 2023-04-16:
Seems to be broken. I was checking Prusa Slicer Alpha and it is broken in that too. I use this a lot to print sequentially different height objects. It works in Prusa slicer stable release and super slicer. Seems to be broken in Orca, Prusa Alpha.

LukeAllen, 2023-06-23:
> I just want to print the letters in Red and the outer line in White, Letters first, Line Second... Help

For that particular need (which I also had) a workaround is to print it as two separate prints -- print 0.2mm tall white letters with no brim/skirt, then leave them on the build plate, swap to red filament, then print the red part as a separate print. (In Prusa Slicer, I first "Split" my multi-part object into separate objects, then hid everything except the letters, sliced and printed, then hid the letters but not background, sliced and printed again.)

As a refinement: the "letters" object can be 2 or 3 thin layers so the lines are less visible. Just make an extra-thick first layer for the "background" object so they end up the same height.

bionicbone, 2023-06-26:
That's exactly what I did in the end, it worked really well

LukeAllen, 2023-07-06:
I just found an even better workaround for two-color signs -- this post shows how to set up manual multi-material printing: https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles/simple-way-to-print-in-real-multi-color-without-mmu-or-layer-height-based-changes/
With that setup, Prusa Slicer knows it's all 1 object, won't complain about collisions, and will prompt you to do the filament change.

rizwansarwar, 2023-10-28:
Yes but it depends on how much colors you have, for 5 colors over 3 layers, you will need to change filament 15 times. Super slicer allows Sequential mode and allows to look the other way for collisions as long as you take responsibility.

pyro9, 2023-10-29:
I have a set of patches that do that for PrusaSlicer in my repo. With an mk8 nozzle, I have been able to do inlays 3 layers high without issue. I did the patch set since I got tired of having to trick the slicer into doing my bidding. I made the indicator in the GCodePreview a silhouette of my nozzle to help with manual collision detection. I did a couple PRs of the work in progress but heard only crickets.

tyler555g, 2024-06-28:
bump because serial object printing is still borked.

MAJCcustoms, 2025-02-23:
Still need a fix for this

DejitaruJin, 2025-03-06:
Still an issue in the latest beta. It would be one thing if this were a fringe, "trust me I'm an expert with my own machine" type deal, which I have encountered before. But even its visual representation acknowledges that, no, at no point is the hotend going to enter the physical space of another object, even if I _weren't_ being careful about print order. The warning is appreciated in some instances, but disabling slicing because of something so unreliable is not.

closed 2026-09-29 by prusaslicer-bot[bot]