prusa3d/PrusaSlicer #602: Filament Temps being ignored & Print and preview layers are different

jsilva-iii, 2017-12-02:
### Version
Slicer 1.38.3 - win64

rev-parse HEAD`_

### Operating system type + version
win 10 64

### Behavior
- Printing with PVA & PLA temps should be 1st layer 210 & 195, rest 195 & 190 - first layer all temps are 210, afterwards all are 195
- Printing with soluble interface, alternate support layers are printed with PVA and PLA instead of only where needed
* _Steps needed to reproduce the problem_
 * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
NO
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[Speaker_bad.zip](https://github.com/prusa3d/Slic3r/files/1524332/Speaker_bad.zip)
[Slic3r_config_bundle.zip](https://github.com/prusa3d/Slic3r/files/1524333/Slic3r_config_bundle.zip)

bubnikv, 2017-12-03:
In both your config bundle and G-code, you have a PLA filament selected for
all four extruders. You should select the PVA filament for the 4th extruder
to get the soluble interfaces.

If you still have a problem, please attach your geometry file (best to
attach an AMF file for MM prints, go to Platter->export as AMF).

jsilva-iii, 2017-12-04:
Hi, you are right, I regenerated the gcode and it was behaving as expected - was user issue, my appologies.

closed 2017-12-04 by jsilva-iii