prusa3d/PrusaSlicer #603: Changing any setting results in application shutdown

codesnippy, 2017-12-03:
### Version
Slic3r-1.38.3-prusa3d-linux64-full-201712012144

### Operating system type + version
Debian GNU/Linux 9 (stretch)

### Behavior
- Changing any setting check-box or otherwise in the "Printer Settings", "Filament Settings" or "Printer Settings" results in the application closing.

I start Slic3r via shell.
```
> ./Slic3r-1.38.3-prusa3d-linux64-full-201712012144.AppImage
```
I get the following console message.

```
Gtk-Message: Failed to load module "canberra-gtk-module"

 ** (slic3r.pl:5135): WARNING **: invalid source position for vertical gradient
```

Slic3r seems to work like normal as long as I don't change any setting.
If change any settings the application closes and the console returns 

```
Segmentation fault
```
If I run older versions of Slic3r in the same way I don't get this.

bubnikv, 2017-12-04:
I don't see this on Ubuntu 16.04.1

Would you please zip and send me the content of your ~/.Slic3r directory with all your presets and slic3r.ini?

Thanks

codesnippy, 2017-12-04:
Hereby I send you the .Slic3r folder as requested,

With kind Regards,
Wouter

bubnikv, 2017-12-05:
Sorry, the file did not make it through.

codesnippy, 2017-12-05:
I have placed the file here http://www.codesnippy.com/Slic3r.zip please notify me as I can remove it again.

bubnikv, 2017-12-05:
[Slic3r.zip](https://github.com/prusa3d/Slic3r/files/1532630/Slic3r.zip)

bubnikv, 2017-12-05:
I have attached it, you may remove it from codesnippy.

codesnippy, 2017-12-05:
Thank you. Hope this helps

bubnikv, 2017-12-05:
Funny thing, it crashes on me even on Linux if I use your configuration files. Somehow the new C++ configuration layer does not like your configuration setup. I am looking into it.

codesnippy, 2017-12-05:
Strange thing is the old version's did not have this problem. I use the same for the old version's

bubnikv, 2017-12-05:
I have rewritten the configuration layer from Perl to C++, so I may have introduced new errors. I think I have found the logical error actually, it only happens if you start your profile names with a character with a lower ASCII value than '-' :-) I will fix that tomorrow.

codesnippy, 2017-12-05:
thanks. Thats some quick bug hunting you're doing ;-) Looking forward to the coming release

bubnikv, 2017-12-06:
Fixed with https://github.com/prusa3d/Slic3r/commit/75dcdb84b1996a2ae507643042df504e1cd5d056.
Thanks for the report. Closing.

closed 2017-12-06 by bubnikv