prusa3d/PrusaSlicer #604: Missing Polycarbonate

AndWik, 2017-12-03:
### Version
_Version of Slic3r used goes here_
1.37.2 Win 64
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Pro 64-bit
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Missing PolyCarbonate as Filament type
In Filament properties it is missing
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-12-03:
We have a print profile for the Polymaker PC-Max
https://github.com/prusa3d/Slic3r-settings/blob/master/Slic3r%20settings%20MK2/filament/Polymaker%20PC-Max.ini
https://github.com/prusa3d/Slic3r-settings/blob/merged/Settings/filament/Polymaker%20PC-Max.ini

bubnikv, 2017-12-04:
Does the filament profile satisfy your request?

AndWik, 2017-12-04:
Hi

 

Thanks but it still says: filament_type = ABS

 

I assume it is useful anyway

 

Thanks

Mr Anders Wikström
Sweden

 

https://github.com/prusa3d/Slic3r-settings/blob/master/Slic3r%20settings%20MK2/filament/Polymaker%20PC-Max.ini
https://github.com/prusa3d/Slic3r-settings/blob/merged/Settings/filament/Polymaker%20PC-Max.ini

bubnikv, 2017-12-04:
> Slic3r%20settings%20MK2/filament/Polymaker%20PC-Max.ini
> Settings/filament/Polymaker%20PC-Max.ini
> AMWmpoPOLSsTdHV0ZieA6yKcb-jpS4S1ks5s8wxAgaJpZM4Qzt9i> . <

The filament_type = ABS is for the multi-material wipe tower to pick the
correct filament exchange sequence, so we are using an ABS multi-material
profile for PC.
One day we will have the filament exchange sequences configurable, but for
now I think it is correct.

AndWik, 2017-12-04:
Hi again

 

Thank you for your quick response. It is a great software and I love all your work.

 

Best regards

Mr Anders Wikström

bubnikv, 2017-12-05:
Thanks, closing.

closed 2017-12-05 by bubnikv