prusa3d/PrusaSlicer #605: 1.38.3 Custom filament G-code not writing to output .gcode

tjcolejr, 2017-12-03:
### Version
1.38.3 64bit

### Operating system type + version
Win 10 64bit  MMU in single mode

### Behavior
If I enter a custom M900 Kxxx value in the start G-code specific to a filament under "Filament Settings," it is not being sent to the output .gcode file.  It does list it at the end of the .gcode file where all the settings are commented.  I have to enter it into the Start G-code under "Printer Settings" for it to be written to the .gcode file.

This used to work. in single mode, but I'm not certain about MM mode

bubnikv, 2017-12-04:
Thanks, I will look into it.

I will likely change Slic3r to provide a "has_wipe_tower" boolean variable to the g-code parser and we will re-add the priming line to the start G-code through a conditional expression.

bubnikv, 2017-12-05:
This has been fixed with https://github.com/prusa3d/Slic3r/commit/2b0b8e6e688fc60c79edb3f1282b7e06b60dfc17. Thanks for the report.

closed 2017-12-05 by bubnikv