prusa3d/PrusaSlicer #606: Inner thread sliced as cylinder even though it shows in preview and slices into correct shape under different orientation

myriadhero, 2017-12-04:
### Version

1.37.2-prusa3d-win64

### Operating system type + version
windows 10 version 1709 build 16299.64

### Behavior

Please see attached screenshots and model

The model is a simple ball with a cut-off that has 18mm diameter hole with a metric thread, the model was made in Autodesk Inventor.

The model shows fine in the preview with all the threads, when imported its on its side and if i slice it like that, the threads are reflected in the slicing.

When I stand it flat, so the hole is vertical, slicing produces a vertical straight walled inner hole rather than a thread.

I have tried varying thread in Inventor, also exporting it as .obj, restarting computer, all sort of different settings in slic3r...

I have mating outer threaded model that slices properly. I don't remember if I previously printed a model with internal thread, but I never had issues with outside threads. This is weird.

* _Steps needed to reproduce the problem_

import model, rotate so the hole is vertical, slice with any settings

  * _If this is a command-line slicing issue, include the options used_

* _Expected Results_

spiralling thread

* _Actual Results_

straight walled cylinder 

  * _Screenshots from __*Slic3r*__ preview are preferred_

please see attached

_Is this a new feature request?_

no

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[image]
[image]
[image]
[slic3rBugReport.zip](https://github.com/prusa3d/Slic3r/files/1527006/slic3rBugReport.zip)

myriadhero, 2017-12-04:
Here's the config.ini, forgot to attach

[config.zip](https://github.com/prusa3d/Slic3r/files/1527065/config.zip)

bubnikv, 2017-12-04:
I think it is one of the https://github.com/prusa3d/Slic3r/labels/model%20import issues.

Your model consists of two bodies: The ball and the thread. For Slic3r to be able to process it correctly as of now, you need to merge the two bodies into one.

[image]

myriadhero, 2017-12-04:
Ahh ok, got it, just got confused because outer thread generated same way always works fine and it also works fine in another orientation...

Thanks, will keep it in mind

closed 2017-12-04 by myriadhero

FidelCapo, 2018-05-11:
Tested and fixed in Slic3rPE-1.40.0-alpha that will be released soon.