prusa3d/PrusaSlicer #607: Multimaterial printer switches filament at the  wrong time during a print

dhjohnston66, 2017-12-05:
### Version
1.37.2-prusa3d

### Operating system type + version
macOS 10.13.1

### Behavior
* The incorrect extruder is selected during a two-material multi-material print

_Steps to reproduce the issue_
* Created multi-material model in Slic3r from two STL files

_Expected Result_
* Model with blue 'atom' in center, yellow on outside
[image]

_Actual Result_
* Center started with blue, but then switch to yellow, outside ended up blue
[image]

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[ptcl4.zip](https://github.com/prusa3d/Slic3r/files/1529597/ptcl4.zip)

bubnikv, 2017-12-05:
Hello Dave. 

I checked your G-code in Simplify3D G-code viewer and I don't see this effect. I tried to generate the same G-code from the data (config and AMF) you provided. I loaded the generated G-code into the Simplify3D G-code viewer and everything seems to be all right. 

As a last thing, I will print the G-code generated by you on a MM printer. If that works, then there must be a problem on your side. If it fails, then there is likely an issue both in Slic3r and a Simplify3D G-code viewer. As of now, I tend to think that the first assumption is correct. May it be, that you were printing an old version of your G-code?

Vojtech

bubnikv, 2017-12-05:
I  have printed the G-code you have provided and I am getting the correct result. The only problem I had was a lifting of one of the balls.

[image]

bubnikv, 2017-12-05:
I wonder whether your issue has something to do with the Slic3r 1.37.1 release bug https://github.com/prusa3d/Slic3r/issues/492, which has been corrected in Slic3r 1.37.2.

dhjohnston66, 2017-12-06:
OK - I can confirm that what I uploaded yesterday prints correctly. But with my next print (different model) the same problem occured.

I've regenerated the files, tested the print, and now the problem is happening again. I am not sure what is different. I have attached the relevant files.

[ptcl4-v2.zip](https://github.com/prusa3d/Slic3r/files/1533626/ptcl4-v2.zip)

bubnikv, 2017-12-06:
Thank you, now I am able to reproduce your issue even with the current beta. Now that I am able to reproduce the issue, I can fix it.

bubnikv, 2017-12-06:
It seems to me the issue is connected to the raft, which is rarely used, yet less with the multi-material setup. Indeed, if you check the G-code in some G-code viewer, you will find out, that there is the 3rd layer  missing at the wipe tower. The missing wipe tower layer is certainly a bug, but it eludes me how it could mess up the extruder assignment of the rest of the print. I hope to track that tomorrow.

m0n5t3r, 2017-12-07:
unless I'm mentally rotating the photo wrong, it looks like it's missing the last tool change and wipe tower layer (peripheral balls are smaller than the center ball, and the wrong colour starts after the smaller balls are done)

jjakubovic, 2017-12-08:
I have the same problem. I have 4 material model that requires raft (otherwise its thin legs fall over during print) and as soon as the extruder switching starts, the incorrect ones are selected.

I've managed to fool the slicer by lifting the object a bit (and adding a small cube on the ground to prevent settling) and turning off the raft. Generated supports substituted the raft. But Slic3er fix would be handy.

I'm on Slic3er 1.37.2. and Firmware 3.1.0.
I can supply the files if needed (but you probably have the repro already).

Thanks

bubnikv, 2017-12-08:
I am looking into it, it will be fixed in the next beta.

bubnikv, 2017-12-11:
Here is an OSX continuous integration build.

[Slic3r-1.38.3.23-prusa3d-full-g61e6f23-201712111622.zip](https://github.com/prusa3d/Slic3r/files/1548690/Slic3r-1.38.3.23-prusa3d-full-g61e6f23-201712111622.zip)

Please note that we have moved the user presets from a Slic3r to Slic3rPE directory, and that the print/filament/printer profiles are now stored in Slic3rPE/presets/ directory, not in Slic3r/
If you just start this Slic3r, you will be welcomed with a wizard offering bundled profiles for our printers :-)

dhjohnston66, 2017-12-13:
I tried out the build from the link above and I'm still seeing the same issues. Gcode, AMF, config files attached in zip file. It's colorful, but not what was intended.

[image]

[boh3.zip](https://github.com/prusa3d/Slic3r/files/1554103/boh3.zip)

bubnikv, 2017-12-13:
Thanks for the data. This time the bug was caused by the famous unreliability of comparing two floating points in a computer. Very often the floats are getting slightly different due to numeric rounding, and the compare operation returns false.

Please try the following continuous integration build. It should be working now. Also in this build I have fixed another bug which was slipping under radar for a long time: The priming towers were extruded at the layer height 0.15 dictated by the firmware, which caused some extruder skipping. I have fixed that and the priming towers are now extruded at the 1st layer height. I did not test the priming towers bugfix on a printer yet, so proceed with care :-)

[Slic3r-1.38.3.32-prusa3d-full-g08b74f8-201712131436.dmg.zip](https://github.com/prusa3d/Slic3r/files/1555855/Slic3r-1.38.3.32-prusa3d-full-g08b74f8-201712131436.dmg.zip)

bubnikv, 2017-12-14:
Fixed in 1.38.4. Closing.
Thanks for all the relevant data.

closed 2017-12-14 by bubnikv

dhjohnston66, 2017-12-15:
Works beautifully. Thanks!