prusa3d/PrusaSlicer #608: Cmake doesn't respect perl's system library locations

smellyfis, 2017-12-05:
### Version
1.37.2 -1.38.3 possibly newer

### Operating system type + version
gentoo linux

### Behavior
slic3r-prusa doesn't respect system perl library locations since using the cmake driven build system

### Fix
 the following patch will install perl library stuff into the systems perl location

```diff
--- a/CMakeLists.txt	2017-11-22 21:23:00.656956898 -0500
+++ b/CMakeLists.txt	2017-11-22 21:26:34.038774760 -0500
@@ -41,6 +41,7 @@
 message("PATH: $ENV{PATH}")
 message("PERL5LIB: $ENV{PERL5LIB}")
 find_package(Perl REQUIRED)
+find_package(PerlLibs REQUIRED)
 
 add_subdirectory(xs)
 
@@ -53,5 +54,5 @@
 
 file(GLOB MyVar var/*.png)
 install(FILES ${MyVar} DESTINATION share/slic3r-prusa3d)
-install(FILES lib/Slic3r.pm DESTINATION lib/slic3r-prusa3d)
-install(DIRECTORY lib/Slic3r DESTINATION lib/slic3r-prusa3d)
+install(FILES lib/Slic3r.pm DESTINATION ${PERL_VENDORLIB})
+install(DIRECTORY lib/Slic3r DESTINATION ${PERL_VENDORLIB})
--- a/xs/CMakeLists.txt	2017-11-22 21:23:00.656956898 -0500
+++ b/xs/CMakeLists.txt	2017-11-22 21:26:34.038774760 -0500
@@ -549,5 +549,5 @@
 endif ()
 
 # Installation
-install(TARGETS XS DESTINATION lib/slic3r-prusa3d/auto/Slic3r/XS)
-install(FILES lib/Slic3r/XS.pm DESTINATION lib/slic3r-prusa3d/Slic3r)
+install(TARGETS XS DESTINATION ${PERL_VENDORARCH}/auto/Slic3r/XS)
+install(FILES lib/Slic3r/XS.pm DESTINATION ${PERL_VENDORLIB}/Slic3r)
```

smellyfis, 2017-12-05:
attaching patch as a file

[slic3r-1.37.2-cmake_paths.patch.zip](https://github.com/prusa3d/Slic3r/files/1531799/slic3r-1.37.2-cmake_paths.patch.zip)

bubnikv, 2017-12-05:
We still have another issue open, where the migration to CMake has been discussed: https://github.com/prusa3d/Slic3r/issues/420

The Slic3r Prusa Edition should install in a way that it would allow the upstream Slic3r to install as well. Also I am a bit worried that each Linux distro will have the Perl installation set up differently and the simple CMake install build will do more harm than good. I am happy that the distro maintainers package our Slic3r PE appropriately for their distro. I am open to suggestions, but I am worried of simplistic solutions.

smellyfis, 2017-12-05:
The PerlLibs cmake module should find the perl installation or allow the user to overwrite them.

As far as allowing upstream to install as well I think we could abstract the destination directory so that it either overwrites or installs elsewhere. But, if the plan isn't to push stuff upstream we should rename all of the Slic3r instances to something else, eg Slic3rPE.

commit 79ee7c9a (2018-03-05): Fix #608  Credit: Dylan "smellyfis" Thies
closed 2018-03-06 by bubnikv